freifunk-berlin/firmware #313: TP-Link CPE210 v1.1

carsifreifunk, 2015-12-21:
Nun ist eine CPE210 in der Version 1.1 aufgetaucht. Laut Recherche sind die Versionen identisch, jedoch lässt sich die "http://buildbot.berlin.freifunk.net/buildbot/stable/0.1.2/ar71xx/default/openwrt-ar71xx-generic-cpe210-220-510-520-squashfs-factory.bin" nicht hoch laden. Die Fehlermeldung sagt aus, dass die Hardwareversion nicht unterstützt wird. Es soll reichen die Versionsnummer dem Image hinzuzufügen [1].

Beispiel:
static const char cpe510_support_list[] =
"SupportList:\r\n"
"CPE510(TP-LINK|UN|N300-5):1.0\r\n"
+ "CPE510(TP-LINK|UN|N300-5):1.1\r\n"
"CPE520(TP-LINK|UN|N300-5):1.0\r\n"
+ "CPE520(TP-LINK|UN|N300-5):1.1\r\n"
"CPE210(TP-LINK|UN|N300-2):1.0\r\n"
+ "CPE210(TP-LINK|UN|N300-2):1.1\r\n"
"CPE220(TP-LINK|UN|N300-2):1.0\r\n"
- "\r\n";
+ "CPE220(TP-LINK|UN|N300-2):1.1\r\n";

Kann das bitte jemand ins Image bauen.

Vielen Dank

Gruß
Carsten

[1] http://openwrt-devel.openwrt.narkive.com/AAjsjKcs/patch-0-3-add-support-for-cpe210-220-510-520-v1-1

Benjarobbi, 2015-12-21:
Hallo, hast du die von Reichelt?

carsifreifunk, 2015-12-23:
Die ist von Notebooksbilliger.de. Aber auch die von Reichelt scheinen v1.1 zu sein.

Noch eine Ergänzung. Wir haben auch versucht ein normales openWRT Chaos Calmer zu installieren, mit selben Effekt.
Dann haben wir zuerst probiert, auf das Stockrom downzugraden, welchen bei openWRT verlinkt ist und haben auf die neuste Firmware von TP-Link upgegraded. Beides ohne Erfolg. Mit der neusten Firmwareversion, soll die Fehlerbeschreibung besser sein, das Gegenteil war aber der Fall. Mit der niedrigen Version sagt er noch, dass die Hardware nicht unterstützt wird und in der neuen Version sagt er nur Fehler und das Log funktioniert auch nicht mehr.

booo, 2015-12-23:
Ich versuche Support für die Router in die Firmware einzubauen. Etwas Geduld bitte.

@carsifreifunk Was heißt "normales openWRT Chaos Calmer? Könntest du mir einen Link zur Firmware oder eine OpenWRT-Revision sagen?

carsifreifunk, 2015-12-23:
Ich habe Geduld und freue mich, dass du dich der Sache annimst.

Ich meine das Chaos Calmer 15.05, wie es auf der openWRT-Seite zur CPE210[1] verlinkt ist.

[1] https://wiki.openwrt.org/toh/tp-link/tl-cpe210

booo, 2015-12-23:
Mein Kommentar zur Geduld war nicht böse gemeint.

Ihr müsst einen Build aus dem openwrt trunk nutzen. Soweit ich das sehen kann wurde der Patch nur in den trunk gemerged:

http://git.openwrt.org/?p=openwrt.git;a=commit;h=c8315df1affa8eadb2ccbfd421a1e5806a444f47

Wir könnten die OpenWrt-Maintainer fragen, ob sie den Patch auch in chaos_calmer (15.05) einbauen können. Alternativ bauen wir den Patch in unser Build-System ein.

booo, 2015-12-31:
Hat jemand den Patch/Backport getestet und kann grünes Licht für einen Merge geben?

closed 2016-01-01 by booo

booo, 2016-01-01:
Laut Mailingliste funktioniert das Image. Merged.

drakessenyonga, 2016-05-26:
Hello, Anyone had luck with cpe 210 v1.1. openwrt trunk. When i use the openwrt in trunk, no interface comes up not even the ethernet. I can only access it via fail safe. Any suggestions. Thank you.

SvenRoederer, 2016-05-26:
@drakessenyonga : we use OpenWRT ChaosCalmer for the current firmware.
For problems with OpenWRT trunk you better check with the OpenWRT guys

lynxis, 2016-05-27:
@drakessenyonga lede/master works.

drakessenyonga, 2016-05-27:
@lynxis where do i find the lede / master. am absolutely new to firmware. @SvenRoederer i tried chaos calmer too but the device says firmware not supported.

lynxis, 2016-05-27:
http://lede-project.org/
see ftp://ftp.halifax.rwth-aachen.de/lede/snapshots/targets/ar71xx/generic/lede-ar71xx-generic-cpe210-220-squashfs-factory.bin

SvenRoederer, 2016-05-27:
as Freifunk Berlin ses unpatched OpenWRT-CC for the CPE210 (incl. HW V1.1) support, I assume you are missing something in your workflow.

drakessenyonga, 2016-05-29:
Hello, Thank you all for the assistance. I managed to build firmware. 
Thanks all.