freifunk-berlin/firmware #314: Images "default4MB" to big since build #692

SvenRoederer, 2015-12-22:
all 4MB-type images failing to build as they are to big (http://buildbot.berlin.freifunk.net/builders/ar71xx/builds/713/steps/shell_1/logs/stdio)

> [mktplinkfw] kernel length aligned to 1049196
> [mktplinkfw] *** error: images are too big

booo, 2015-12-24:
This is a know problem and one reason why we still have no new release for the firmware.

sarumpaet, 2015-12-24:
One idea was to get rid of iptables (as the fw3 binary can handle netfilter stuff directly). https://github.com/freifunk-berlin/firmware-packages/pull/62 is one part of that, as are the luci-mod-freifunk-ui branches (that have 4MB builds). I need to have PR 62 merged first to continue working on that as otherwise I'd be more rebasing than coding.

Closing this issue as it's redundant and not referring to releases anyways.

closed 2015-12-24 by sarumpaet

SvenRoederer, 2015-12-25:
@sarumpaet you link seems to be wrong, but I assume you are referencing https://github.com/freifunk-berlin/firmware-packages/commits/luci-mod-freifunk-ui?

sarumpaet, 2015-12-26:
1. The link is correct. PR 62 replaces a rather horrible (and defunct) ip-blacklisting hack that is supposed to prevent OpenVPN from connecting to VPN servers via the mesh with a better solution, and removes freifunk-berlin-firewall-defaults' iptables dependency on the side.

2. Yes, the luci-mod-freifunk-ui branch in firmwares-packages, but there's also https://github.com/freifunk-berlin/firmware/tree/luci-mod-freifunk-ui which uses that and plays with some other requirements stuff. For that branch, 4MB packages get build.