freifunk-berlin/firmware #316: no uplink-route for DHCP-clients

SvenRoederer, 2015-12-30:
@stargieg in build 718 (http://buildbot.berlin.freifunk.net/builders/ar71xx/builds/718) (https://github.com/freifunk-berlin/firmware-packages/tree/luci-mod-freifunk-ui) there is no uplink-route for clients on interface "br-dhcp".

"ip rule" is not showing any "prio 19999" rules, so also no rule to "olsr-tunnel".
> ip rule add iif br-dhcp pref 19999 lookup olsr-tunnel
makes routing work

sarumpaet, 2015-12-31:
That branch is by me, not stargieg. :) Thanks for testing! I'm actually surprised that it works at all, as that specific build was more of a test whether the firmware works without both firewall and iptables packages (note the last commit "WIP" :) ). It'd be perfect if you could have a look or help with further testing. :)

SvenRoederer, 2015-12-31:
loooks like I got lost in the names here ...
I also like to see some progress, esp as these chnges help to reduce the size for these damn tiny 4MB-flash routers.
Think, if nobody has concerns to go into this direction for 0.2.0 , I'm happy to help

sarumpaet, 2015-12-31:
https://github.com/freifunk-berlin/firmware-packages/pull/62 (plus PR 66) and the luci-mod-freifunk-ui package are ready for merge I'd say. Fixing this issue can come later, as can any detail work or further improvements (e.g., launching OpenVPN via procd). We should merge PRs more quickly instead of letting them lie around forever, especially if they fix crucial bugs. I'd be interested in how many connections to VPN03 come from inside the mesh at the moment. ;)

booo, 2016-01-01:
It's a shame that the pull request is laying around so long but I don't want to merge it as it is. I will try to comment on this in the pull request.

SvenRoederer, 2016-01-02:
seems that this missing uplink / faulty routing only existes when doing a fresh install. on a upgraded system from 0.1.2 the uplink was ok ("19999 rules were installed")

SvenRoederer, 2016-01-04:
I made a diff over the generated backup-file between a upgraded and fresh installed router. some lines are no real differences, as I used different IP-ranges in the wizard.
Maybe you can spot the relevant difference
<pre>
sven@hypnosis:/tmp/config$ diff -uNr etc_upgrade/ etc_fresh/    
diff -uNr etc_upgrade/config/defaults etc_fresh/config/defaults
--- etc_upgrade/config/defaults 2015-07-02 11:21:27.000000000 +0200
+++ etc_fresh/config/defaults   2016-01-03 16:10:30.000000000 +0100
@@ -1,7 +1,6 @@
 
 config packages 'default'
        option dhcp '1'
-       option firewall '1'
        option network '1'
        option olsrd '1'
        option openvpn '1'
diff -uNr etc_upgrade/config/dhcp etc_fresh/config/dhcp
--- etc_upgrade/config/dhcp     2016-01-02 02:55:51.000000000 +0100
+++ etc_fresh/config/dhcp       2016-01-03 16:20:24.000000000 +0100
@@ -14,14 +14,7 @@
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.auto'
-
-config dhcp 'lan'
-       option interface 'lan'
-       option start '100'
-       option limit '150'
-       option leasetime '12h'
-       option dhcpv6 'server'
-       option ra 'server'
+       option localservice '1'
 
 config dhcp 'wan'
        option interface 'wan'
@@ -29,11 +22,11 @@
 
 config domain 'frei_funk'
        option name 'frei.funk'
-       option ip '10.230.192.1'
+       option ip '10.230.197.209'
 
 config domain 'frei_funk_ipv6'
        option name 'frei.funk'
-       option ip 'fde7:d3d3:80b1::1'
+       option ip 'fdc9:cda0:0e43::1'
 
 config odhcpd 'odhcpd'
        option maindhcp '0'
@@ -49,5 +42,5 @@
        option dhcpv6 'server'
        option ra 'server'
        option start '2'
-       option limit '253'
+       option limit '13'
 
diff -uNr etc_upgrade/config/ffwizard etc_fresh/config/ffwizard
--- etc_upgrade/config/ffwizard 2015-07-02 11:29:59.000000000 +0200
+++ etc_fresh/config/ffwizard   2016-01-03 16:20:24.000000000 +0100
@@ -6,7 +6,7 @@
        option enableStats '0'
        option private_ap '0'
        option ssid 'test.berlin.freifunk.net'
-       option dhcpmesh '10.230.192.0/24'
+       option dhcpmesh '10.230.197.208/28'
        option meship_radio0 '10.31.21.125'
        option runbefore 'true'
 
diff -uNr etc_upgrade/config/firewall etc_fresh/config/firewall
--- etc_upgrade/config/firewall 2016-01-02 02:55:51.000000000 +0100
+++ etc_fresh/config/firewall   1970-01-01 01:00:00.000000000 +0100
[... 431 lines cut ...]
+       option hwmode '11bgn'
        option diversity '1'
        option channel '13'
 
diff -uNr etc_upgrade/crontabs/root etc_fresh/crontabs/root
--- etc_upgrade/crontabs/root   2015-07-02 11:30:00.000000000 +0200
+++ etc_fresh/crontabs/root     2016-01-03 16:20:28.000000000 +0100
@@ -1,4 +1,4 @@
-12 * * * *     /usr/sbin/owm.lua
+26 * * * *     /usr/sbin/owm.lua
 */5 * * * *    killall -HUP dnsmasq
 */5 * * * *    /usr/sbin/ff_olsr_watchdog
 0 6 * * * ifup wan
Binärdateien etc_upgrade/dropbear/dropbear_dss_host_key und etc_fresh/dropbear/dropbear_dss_host_key sind verschieden.
Binärdateien etc_upgrade/dropbear/dropbear_rsa_host_key und etc_fresh/dropbear/dropbear_rsa_host_key sind verschieden.
diff -uNr etc_upgrade/shadow etc_fresh/shadow
--- etc_upgrade/shadow  2015-07-02 11:29:05.000000000 +0200
+++ etc_fresh/shadow    2016-01-03 16:19:26.000000000 +0100
@@ -1,4 +1,4 @@
-root:$1$HvqSMuKv$rxqQxt546CJcmoLwJobm4/:16618:0:99999:7:::
+root:$1$xWecZWws$x1oe8J0EfOoDmbqebV5171:16803:0:99999:7:::
 daemon:*:0:0:99999:7:::
 ftp:*:0:0:99999:7:::
 network:*:0:0:99999:7:::
</pre>

sarumpaet, 2016-02-07:
Let's keep this in mind. Closing this as we typically don't create issues for bugs in dev-only builds.

closed 2016-02-07 by sarumpaet