freifunk-berlin/firmware #326: TP-Link CPE210

Ainadilion, 2016-02-05:
Beim Versuch die entsprechende Datei im Menü einzuspielen, erfolft folgende Fehlermeldung: The hardware version is not supported. Wie kann der Router geflasht werden?
mfg Steffen

SvenRoederer, 2016-02-05:
Welche Firmwareversion willst du flashen?
auf welche Hardwareversion willst du flashen?

Hardwareversion V1.1 ist bisher nur in kathleen-0.2.x unterstützt

Ainadilion, 2016-02-06:
Hallo,

derzeit ist folgende Firmware auf dem Gerät: 1.3.0 Build 20150612 Rel. 50957
Ich versuche diese zu
installieren: openwrt-ar71xx-generic-cpe210-220-510-520-squashfs-factory

mfg
Steffen

booo, 2016-02-06:
@Ainadilion Wo kommt die Firmware her, die du flashen willst?

Wahrscheinlich hast du Versin 1.1 des Routers und der wird nur im [unstable branch](http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/761/default/) unterstützt.

Ainadilion, 2016-02-06:
Hallo, hatte mir die Firmware von der Berliner Freifunkseite
heruntergeladen, muss ich bei der Version im Link etwas Besonderes beachten?

mfg Steffen

SvenRoederer, 2016-02-06:
welche Hardwareversion du hast, steht neben dem kleinen aufkleber mit der S/N

Ainadilion, 2016-02-07:
Hi, die Version lautet Ver: 1.1.
mfg Steffen

sarumpaet, 2016-02-07:
Siehe https://wiki.freifunk.net/Berlin:Firmware#Router-spezifische_Hinweise , der v1.1 wird von den aktuellen Releases noch nicht unterstützt, booo hatte bereits einen Link zu Entwicklungs-Images geschickt.

closed 2016-02-07 by sarumpaet

Ainadilion, 2016-02-07:
Hallo, also macht es Sinn noch etwas zu warten?

mfg Steffen

sarumpaet, 2016-02-07:
Das verlinkte Image kannst Du gerne ausprobieren, das läuft bei anderen Leuten problemlos.

Ainadilion, 2016-02-07:
Hallo,

hab jetzt das Image eingespielt, jedoch gibt es keine Verbindung über
*http://192.168.42.1/
<http://192.168.42.1/>. Gibt es hier auch Besonderheiten zu beachten für
die Konfiguration?*

*mfg Steffen*

Ainadilion, 2016-02-07:
Hallo, Kommando zurück, jetzt geht es, musste nur auf automatisches
Beziehen der IP-Adresse zurückschalten ;-)

mfg Steffen

SvenRoederer, 2016-02-07:
hast du schonmal die andere Netzwerkbuchse probiert?

sonst meld dich mal auf der Mailingliste, da lesen mehr leute, die praktische Erfahrungen haben.