freifunk-berlin/firmware #337: don't build openwrt SDK

booo, 2016-03-09:
At the moment we build the openwrt SDK on each build. I never used the SDK and as far as I know it's also no dependency for something we need. I suggest that we don't build the SDK anymore.

SvenRoederer, 2016-03-10:
Atm. the only reason for the SDK might be, to use it to build the tunnelbroker for the VPN experiments.

booo, 2016-03-10:
Who does this?

SvenRoederer, 2016-03-10:
Who knows ... I can just imagine this as a valid reason for using the SDK.

Maybe it's a good idea to stop not build the SDK in unstable build and only in the final stable build. So in case anyone likes to compile something for released stable, he can use the SDK for this.

booo, 2016-03-10:
Have you ever tried to build something with the SDK? Do you know if it's really working? If they want to build something they can also use the normal make process.

SvenRoederer, 2016-03-22:
what about building the SDK only on stable releases? so interested people can just use it without compiling all themselves?

This requires just a test in the Makefile for the build-type stable/unstable (depending on the release-name, as done by the build-bots), and the Makefile  can change this option in openwrt/.config

FreifunkUFO, 2016-05-19:
SDK is needed when compiling/ updating packages.. and for freifunk-berlin-unknown packages! so please build SDK with your firmwares!

andrenarchy, 2016-05-19:
I also use the SDK sometimes. So +1 for keeping it. :)

booo, 2016-05-19:
Can someone outline the process of building something with the sdk? If it really works we could maybe improve the whole build process with this.

lynxis, 2016-05-29:
+1 for keeping it.
@boo see lede-project use it in their 2 phase build. first build only the base system and second build all packages using the sdk.
https://git.lede-project.org/?p=buildbot.git;a=blob;f=phase2/master.cfg;h=e61fce55b5dd6d0f5f2ea38a1b15fbb91e05d995;hb=HEAD

FreifunkUFO, 2016-05-29:
@booo please see http://wiki.openwrt.org/doc/howto/obtain.firmware.sdk
and try it, f.i. with umurmur-openssl or cjdns..

closed 2016-08-08 by lynxis