freifunk-berlin/firmware #338: OpenVPN crashes on the TP-Link 841n v10

philiplb, 2016-03-14:
Hi,
I wanted to get into Freifunk and setup a TP-Link 841n v10 with the latest build I could find: kathleen 0.2.0-alpha+424da9d
I just did the setup wizzard and changed nothing else.
So far, everything seems to work but OpenVPN. It doesn't start and so internet isn't working as well.
This line from the log seems interesting:

> deamon.info procd: Instance openvpn: instance 1 s in a crash loop 6 crashes, 0 seconds since last crash

It is this router:
http://openwifimap.net/#detail?node=PhilipLB.olsr

Any clue? :)
Any more information you would find usefull?

SvenRoederer, 2016-03-14:
gibt's von OpenVPN irgendwelche Meldungen?
der procd sagt ja nur, dass openvpn immer abstürtzt / nicht startet

SvenRoederer, 2016-03-15:
Ich meine, wir haben dieses Image auch gestern in der Regugees-unterkunft "Groß-Berliner-DAmm" installiert (cjd-1, cjd-2) und da läuft es bisher.
Dort werkeln TPlink WR842.

philiplb, 2016-03-15:
Ich hab hier mal einen Gist vom System Log und Kernel Log:
https://gist.github.com/philiplb/f789d3ee463147a1fdbc
Gibt es noch mehr Orte/Logs, die hier interessant wären?

SvenRoederer, 2016-03-15:
das system-log reicht erstmal ...
aber ich dem o.g. Ausschnitt finde ich nichtmal, die von dir erwähnte ursprüngliche fehler-meldung

philiplb, 2016-03-15:
Ah, stimmt, da wurde wohl ein wenig in den Logs gerollt. Ich habe mal neu gebootet und hier ist der wohl interessante Ausschnitt aus dem System Log:

> Tue Mar 15 23:30:31 2016 user.notice ff-userlog: WAN interface is up, starting OpenVPN
> Tue Mar 15 23:30:31 2016 user.notice ff-vpn-hotplug: Using UCI OpenVPN configuration
> Tue Mar 15 23:30:32 2016 daemon.err openvpn(ffvpn)[2682]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:32 2016 daemon.warn openvpn(ffvpn)[2682]: Use --help for more information.
> Tue Mar 15 23:30:37 2016 daemon.err openvpn(ffvpn)[2688]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:37 2016 daemon.warn openvpn(ffvpn)[2688]: Use --help for more information.
> Tue Mar 15 23:30:42 2016 daemon.err openvpn(ffvpn)[2689]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:42 2016 daemon.warn openvpn(ffvpn)[2689]: Use --help for more information.
> Tue Mar 15 23:30:47 2016 daemon.err openvpn(ffvpn)[2690]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:47 2016 daemon.warn openvpn(ffvpn)[2690]: Use --help for more information.
> Tue Mar 15 23:30:52 2016 daemon.err openvpn(ffvpn)[2691]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:52 2016 daemon.warn openvpn(ffvpn)[2691]: Use --help for more information.
> Tue Mar 15 23:30:57 2016 daemon.err openvpn(ffvpn)[2698]: Options error: You must define TUN/TAP device (--dev)
> Tue Mar 15 23:30:57 2016 daemon.warn openvpn(ffvpn)[2698]: Use --help for more information.
> Tue Mar 15 23:30:57 2016 daemon.info procd: Instance openvpn::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

SvenRoederer, 2016-03-16:
das hab ich auch gesehen ...
kommt von https://github.com/freifunk-berlin/firmware-packages/tree/openvpn-hotplug (https://github.com/SvenRoederer/freifunk-berlin-firmware-packages/commit/4d7545e4f1dc08f6e2a1324a0a56a2346b46f772) da beim Neuaufsetzen das /etc/config/openvpn noch leer ist aber vom ffwizzard auf "enabled" gesetzt wird (https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/shareInternet.lua#L66) und dann gibt's die probleme.

--> erstmal uci -> openvpn -> ffvpn -> enabled auf "0" setzen, dann wird die hardcoded config genommen

philiplb, 2016-04-02:
Heute auf diese Version aktualisiert:
http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/866/default_4MB/kathleen-0.2.0-alpha%2Bfebeed9-tl-wr841-v10-sysupgrade.bin

Leider das selbe Symptom, OpenVPN kommt nicht hoch.

SvenRoederer, 2016-04-02:
hast du wirklich aktualisiert?
Wenn ja, mach bitte ein neuinstallation (ohne übernehmen der alten Einstellungen), da das die unstable software ist, wird das upgraden erst später wirklich funktionieren

philiplb, 2016-04-02:
Das war's. Ich habe den Router komplett neu installiert und jetzt funktioniert alles mit dem Build 866. :) Danke.

closed 2016-04-02 by philiplb