freifunk-berlin/firmware #342: opkg-sources URL pointing to invalid location

SvenRoederer, 2016-04-03:
- for ,at least the unstable builds, the opkg-sources pointing to "http://buildbot.berlin.freifunk.net/buildbot/stable/0.2.0-alpha+4f56084/ar71xx/packages/..." and installation fails on the router
- the Makefile needs a way to include the buildnumber in the default URL ...

lynxis, 2016-05-29:
@SvenRoederer it's broken since https://github.com/freifunk-berlin/firmware/commits/1e75bbac
it's your commit ;). It broke it becaue the buildbot try to replace it.
https://github.com/freifunk-berlin/buildbot/blob/master/masters/master/master.cfg#L129

lynxis, 2016-05-29:
but the replacement failed because it's no long part of the arch specific configurations.

lynxis, 2016-05-29:
https://github.com/freifunk-berlin/buildbot/pull/34

SvenRoederer, 2016-06-30:
fixed by https://github.com/freifunk-berlin/buildbot/commit/d8a651026190ff9e88b9153a9bc302de945c01f1

closed 2016-06-30 by SvenRoederer