freifunk-berlin/firmware #343: Community setting not working

buch8, 2016-04-10:
0.2.-alpha+a3dd257

Login
Freifunk -> Basic Settings
Community is set to "Freifunk 31c3"!?

After setting Community to Berlin the welcome page looks funny. Resetting to "Freifunk 31c3" does not help. Only going through Freifunkassistent again makes things normal.

SvenRoederer, 2016-04-10:
have you seen this on a fresh of upgraded installation?

buch8, 2016-04-11:
I did a sysupgrade installation from original openwrt factory firmware ( GL-AR150 )

SvenRoederer, 2016-04-11:
There is no guarantee that migration will work outside of the Freifunk-Berlin releases.
Maybe this caused the issue you see,

buch8, 2016-04-11:
I will test with a WR841N.

buch8, 2016-04-13:
OK, looks like default_4MB doesn't have community settings. I'm closing this until I can test with a default factory image.

Minimal kathleen-0.2.0-alpha+265ee70 nearly bricked my WR841N by the way. ( had to use tftp recovery )

SvenRoederer, 2016-04-14:
indeed thes pages are left out to safe fome flash on these routers

What was the problem you experienced with the "minimal" image? Never checked it out for myself yet

buch8, 2016-04-14:
Maybe it is a feature of "minimal" that there is no web interface? I got a WLAN connection but no LUCI on WLAN or LAN. I could connect with ssh but could not log in.
I could recover with the special recovery mode of the WR841N so I didn't need to attach a serial connection. If you have ideas for more research without the need for a serial connection I will try again.

SvenRoederer, 2016-04-14:
Right, no web-interface - ssh not accepting login as of unset password

It's like pure OpenWRT-trunk images: telnet will work

buch8, 2016-04-14:
Ah yes, I noticed later that I didn't try telnet. So it's really minimal.
Sorry, I shouldn't have tried it blindly but it was late...

SvenRoederer, 2016-04-23:
@Mask66 seems also to have seen this

eins ist mir noch bei den Kathleen 2.0-Builds aufgefallen, unabhängig vom Router: sobald man nach der Einrichtung über den Wizard in die Basic Settings geht, steht da vorausgewählt als Profil Freifunk 31c3, egal was man bei der Einrichtung gewählt hat. Wenn man jetzt irgendwas ändert (zB das Profil auf Berlin) und speichert, steht auf der Startseite an der Stelle, wo die Community steht ("Hello and welcome in the network of XXX") nur "/etc/config/profile_berlin" oder sowas in der Richtung. Ist wahrscheinlich nur kosmetisch, aber dennoch vllt erwähnenstwert.

SvenRoederer, 2016-04-24:
please test with current build of branch "unstable_SAm0815" for ar71xx this should become build#889

buch8, 2016-04-24:
build 890 with WR1043ND V2
sysupgrade with configuration results in a welcome page with /etc/config/profile_berlin
profile page gives error, profile(expert) shows nothing
freifunk assistent not working

sysupgrade without config
basic setting works but freifunk assistent gives error after next on first page:

```
/usr/lib/lua/luci/dispatcher.lua:360: Failed to execute cbi dispatcher target for entry '/admin/freifunk/assistent/generalInfo'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/uci.lua:65: bad argument #1 to 'n' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:360: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:135: in function </usr/lib/lua/luci/dispatcher.lua:134>
```

SvenRoederer, 2016-04-25:
tried a different way in build#891 - can you test this please

buch8, 2016-04-25:
build 891 looks better

sysupgrade with config:
welcome ok
Basic system settings ok
Profile + Profile ( Expert ) ok

Change to Freifunk Demo ok
Change back to Freifunk Berlin ok
Freifunk Assistent still working

Mask66, 2016-04-27:
Just tried build 901 sysupgrade without config - same bug, Freifunk 31c3 pre-chosen as profile in the settings after first configuration, on changing the basic settings "/etc/config/profile_berlin" on the welcome page.

Mask66, 2016-04-27:
Tested build 891 - i can confirm there it's fixed, some something between 891 and 901 broke it.

SvenRoederer, 2016-04-27:
did you do a fresh install? when you did sysupgrade with keeping configs, there is a chance, you still have a faulty config-file.

post / check /etc/config/freifunk --> section "system"

Mask66, 2016-04-27:
I did an sysupgrade without keeping the config, reproduced the behaviour on Builds 901, 894, 893 (twice). The bug is reproducible fixed on Build 891. Could it have something to do with https://github.com/freifunk-berlin/firmware/commit/3516070ef462be58c2d17166da6b88cbe6eb5cb7 ? Reverting the change fixing this issue, if i understood it correctly?

buch8, 2016-04-27:
Yes, I see it again too with build 903.

SvenRoederer, 2016-04-27:
build 891 was the last of the "unstable_SAm0815" series ... but had nothing to fix this issue ...
the fix came with build 894, but there was also some changes in upstream-LUCI, which were not part of the "unstable_SAM0815" builds before.
Will recheck here

SvenRoederer, 2016-04-27:
got it ... we changed from "luci-mod-freifunk" to our fork "luci-mod-freifunk-ui" in https://github.com/freifunk-berlin/firmware/commit/bd51bfe9239b8f3bbb815b9d5f4fd42cc0e23473.
But the patch to fix the community-setting (https://github.com/freifunk-berlin/firmware/commit/7ea6789c0300b1495ecc5c74f90f2085185dcf31) was not ported into it.

closed 2016-04-28 by SvenRoederer

SvenRoederer, 2016-04-28:
please retest with build 904

buch8, 2016-04-28:
904 looks good again after sysupgrade with configuration and after a reset of configuration too.

Mask66, 2016-04-28:
I can confirm build 904 looks good.