freifunk-berlin/firmware #346: Archer C7 v2 no 5GHz Ad-Hoc on Builds 881,885

Mask66, 2016-04-21:
Hi,

After flashing the latest builds (881,885) to my Archer C7 v2 and configuring it for a freifunk node I noticed that the 5GHz ad-hoc interface is not coming up.

Looks like it was mentioned in #277 , but without a real resolution, and i couldn't find the older builds mentioned there to test.

Here's how it looks in the config:

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option path 'pci0000:01/0000:01:00.0'
        option disabled '0'
        option country 'DE'
        option doth '0'
        option hwmode '11acn'
        option diversity '1'
        option htmode 'HT20'

Here how it looks in Luci: (notice the "0.000 GHz" and "Country: 00")

Status
	[Signal: 0 dBm / Noise: 0 dBm]
0% 	Mode: Ad-Hoc | SSID: intern-ch36.freifunk.net
BSSID: 02:36:CA:FF:EE:EE | Encryption: -
Channel: 36 (0.000 GHz) | Tx-Power: 0 dBm
Signal: 0 dBm | Noise: 0 dBm
Bitrate: 0.0 Mbit/s | Country: 00

Changing htmode to "HT40", "VHT40", "HT40-", as per https://dev.openwrt.org/ticket/18268 doesn't fix it, adding "option noscan 1" doesn't change anything either.

DIsabling the ap interface doesn't bring up the ad-hoc interface either, as reported in some cases with other routers.

Here's some (maybe) relevant stuff from dmesg:

[   13.070000] ath10k_pci 0000:01:00.0: pci irq legacy interrupts 0 irq_mode 0 r                                                                                        eset_mode 0
[   13.700000] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id                                                                                         0x043202ff sub 0000:0000
[   13.710000] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs                                                                                         1 testmode 1
[   13.730000] ath10k_pci 0000:01:00.0: firmware ver 10.2.4.97 api 5 features no                                                                                        -p2p crc32 f91e34f2
[   13.780000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/                                                                                        hw2.0/board-2.bin failed with error -2
[   13.790000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   13.860000] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store                                                                                        : map pages failed
[   13.870000] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c                                                                                        08
[   14.930000] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 5 htt-op 2 cal file m                                                                                        ax-sta 128 raw 0 hwcrypto  @1

I can happily provide the credentials, if someone wants to tinker with it.

SvenRoederer, 2016-04-21:
are these builds the only builds it fails or just examples? and it fails also with other builds, but you only tested these.
build 881: only has a change to OpenVPN
build 885: has only a different way of applying "patch: ar71xx - patch to fix nondeterministic boot hangs" and changed download -URL for tmux

Mask66, 2016-04-21:
just examples, i just tried build 860 (oldest i could find), same picture

SvenRoederer, 2016-04-21:
just found this on the OpenWRT-devel list: https://lists.openwrt.org/pipermail/openwrt-devel/2016-April/040905.html

maybe it's related.

Mask66, 2016-04-21:
Hm, i did see something about "Country=US" or something like that in the dmesg. But it correctly sets the Regulatory Domain for the 2.4GHz Radio to "DE", also the 5GHz AP part seems to work (correctly set to DE). I'll test if ian ad-hoc 5GHz interface comes up in vanilla OpenWrt.

Mask66, 2016-04-28:
Seems it also happens in Vanilla Openwrt.

booo, 2017-01-24:
Can you post the output from `iw phy0 info`? The supported interface modes should be listed there somewhere.

SvenRoederer, 2017-10-28:
I remember a talk with @elektra42 on last meeting, where we briefly discussed on a Archer C7 V2 and 5GHz.
The chipset is a ath10k, which needs a firmware to be downloaded. But it seems there is no well supported which is supporting AP-mode and IBSS simultaneously. IBSS in general seems to be more or less unmaintained by vendors firmware.

The best seems to be to switch to AP + 802.11s. check https://github.com/freifunk-berlin/firmware/projects/3 for some thoughts

SvenRoederer, 2018-01-04:
there is a posting int the mailinglist, that disabling the adhoc on 5GHz-radio will make the BSS-mode work.
So this indicates that the IBSS-mode is not supported in radio-firmware or that this combo of BSS and IBSS is not supported.

It needs to be checked if BSS and 802.11s will work together. @Mask66 can you check?

Mask66, 2018-02-05:
Sorry for the late reply, all C7 v2 I have are in production, so unfortunately, I can't check. End of the month I should have one here.

SvenRoederer, 2018-02-06:
It takes what it takes :-)

PolynomialDivision, 2018-03-17:
iw says

    Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point

    valid interface combinations:
		 * #{ AP, mesh point } <= 8, #{ managed } <= 1,
		   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

This config works:

    config wifi-iface                             
            option network 'wireless0'   
            option device 'radio0'          
            option bssid '02:36:CA:XXXX'      
            option ssid 'intern-ch36.freifunk.net'
            option mcast_rate '12000'
            option mode 'mesh'           
            option ifname 'wlan0-adhoc-5'  
            option mesh_id 'your-mesh-name'  
            option encryption 'none'
                                    
    config wifi-iface                        
            option ifname 'wlan0-dhcp-5'
            option network 'dhcp'            
            option encryption 'none'              
            option device 'radio0'  
            option mode 'ap'                 
            option ssid 'berlin.freifunk.net'  

What does this mean? OLSRD is using adhoc and not 802.11s?
Can I use 802.11s in my home using olsrd?

PolynomialDivision, 2018-03-17:
Ok just tested:
WDR4300 <- 802.11s (with additional AP) -> Archer C7 V2

Works! :)

I used the config

    config wifi-iface
            option network 'wireless0'
            option device 'radio0'
            option bssid '02:36:CA:XXXXXXXXXE'
            option ssid 'intern-ch36.freifunk.net'
            option mcast_rate '12000'
            option mode 'mesh'
            option ifname 'wlan0-adhoc-5'
            option mesh_id 'freifunk'
            option mesh_fwding '0'
            option encryption 'none'

As descirbed [here](https://lists.berlin.freifunk.net/pipermail/berlin/2015-July/028792.html) u have to set mesh_fwding '0'. Furthermore the email list means we are using option mesh_id 'freifunk'. :)

SvenRoederer, 2018-03-19:
So it's confirmed that this board (and other boards using ath10k) do not like the adhoc-mode but the 802.11s Mesh-mode is working.

bobster-galore, 2018-09-21:
Could we close this because Mesh (11s) is working at 5 GHz on this device?
s.a. https://wiki.freifunk.net/Benutzer:Holger/mesh_(11s)

closed 2018-09-30 by bobster-galore