freifunk-berlin/firmware #347: community-profile still lists 104.0.0.0/8 for mesh_network

SvenRoederer, 2016-04-24:
The current Community-profile for Berlin still references this for meshing.
What is the correct network to announce here?

> config community 'profile'
        option name 'Freifunk Berlin'
        option homepage 'http://berlin.freifunk.net'
        option ssid 'berlin.freifunk.net'
        option ssid_scheme 'addchannelbefore'
        option mesh_network '104.0.0.0/8'
        option splash_network '10.104.0.0/16'
        option splash_prefix '27'
        option latitude '52.52075'
        option longitude '13.40948'

FreifunkUFO, 2016-07-08:
das community profil für berlin ist leider in den normalen repos nicht aktuell, sondern wird erst via patch korrekt gesetzt.
https://github.com/freifunk-berlin/firmware/blob/master/patches/200-update-community-profiles.patch
bitte pflegt das mal ordentlich ein.

SvenRoederer, 2016-07-11:
see also #310 , there was some "veto" on pushing all this upstream

FreifunkUFO, 2016-08-05:
das Ticket ist in englisch und nur schwer zu verstehen :-(

Sinnvolle default-Werte machen "upstream" am meisten Sinn. Nur so funktionieren auf längere Sicht alle Varianten (khalua für Berliner, khalua für nicht-berliner, nicht-kalua (zb lede-meshkit) für berliner, nicht-kalua für nicht-berliner (zb cottbus).

option mesh_network ist mit 104..... extrem veraltet und sollte auf jeden Fall mal geändert werden!

Falls für Khalua noch Patches gebraucht werden, können die ja nach wie vor (in khaluah) verwenden werden.

SvenRoederer, 2018-09-23:
no progress and not obsolete

bobster-galore, 2018-09-24:
To me it seems to be quite easy, replace '104.0.0.0/8' by something working.

BUT there is a lot of historical discussion about fixing upstream or here and what and what not. With a lot of crossreferences back and forth.
May be @booo and @sarumpaet know more about the background and what is necessary to get that solved?

booo, 2018-09-24:
The default value for the mesh_network option is fine. It's 10.0.0.0/8 and our network is a subnetwork of this one.

For reference please take a look how the mesh_network option is used: https://github.com/openwrt/luci/blob/c32097d7d525d0c3e960e558d1c561c9c596bf96/applications/luci-app-meshwizard/luasrc/model/cbi/freifunk/meshwizard.lua#L9

Btw. we don't use this option in our code at all.

booo, 2018-09-25:
I created a change for the upstream repository that got merged:

https://github.com/openwrt/luci/commit/eab986432da4dc52e3293d46a6e0083eb01196fe

closed 2018-09-25 by booo