freifunk-berlin/firmware #351: missing images for ramips

ffcb, 2016-05-02:
hello,

at the actual firmware-builds of kathleen unstable for ramips via http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips several images for different devices (e.g. nexx wt3020) are missing since some builds.

SvenRoederer, 2016-05-02:
checked in upstream OpenWRT: NEXX WT3020 is build in "Default"-profile and "MT7620a"-profile.
Only building "Default" fails to build for our images as of different Flash-size (EX2700 needs 4MB-flavour).
Building "MT7620a" looks fine, but finally errors with Returncode "2" and message
> ( cd /tmp/OpenWrt-ImageBuilder-ramips-mt7620.Linux-x86_64/bin/ramips ; find -maxdepth 1 -type f \! -name 'md5sums'  -printf "%P\n" | sort | xargs md5sum --binary > md5sums )
( cd /tmp/OpenWrt-ImageBuilder-ramips-mt7620.Linux-x86_64/bin/ramips ; find -maxdepth 1 -type f \! -name 'md5sums'  -printf "%P\n" | sort | xargs openssl dgst -sha256 > sha256sums )
make[3]: Leaving directory '/tmp/OpenWrt-ImageBuilder-ramips-mt7620.Linux-x86_64/target/linux/ramips/image'
make[2]: Leaving directory '/tmp/OpenWrt-ImageBuilder-ramips-mt7620.Linux-x86_64'
make[1]: Leaving directory '/tmp/OpenWrt-ImageBuilder-ramips-mt7620.Linux-x86_64'
make: *** No rule to make target 'MT7620a'.  Schluss.

Who has knowledge of the OpenWRT build-scripts?

SvenRoederer, 2016-05-03:
can you check if these images are working?
http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips/221/default/

ffcb, 2016-05-03:
i have updated a active node powered by nexx wt3020 from 0.2.0-alpha+a3dd257 to your provided 0.2.0-alpha+89c77d0 and it seems to work. i will have an eye at that affected node

SvenRoederer, 2016-05-03:
patch added in https://github.com/freifunk-berlin/firmware/commit/56eb03e1688bde57b5183b96ff0f5fd4069e6b1c

closed 2016-05-03 by SvenRoederer