freifunk-berlin/firmware #355: weekly-build includes splash-files

ffcb, 2016-05-12:
hi,

the new weekly-build (Freifunk Berlin kathleen 0.2.0-alpha+530b9a1 ) includes the packages for the splashpage
i used the build for ramips

SvenRoederer, 2016-05-12:
but was'nt it like this all the time?
this package was added to default build in https://github.com/freifunk-berlin/firmware/commit/15cf32455dea597aad79e99c2d01eabdf9a5c944.
and so it made it way also trough the "common-config"-rework (https://github.com/freifunk-berlin/firmware/commit/7ff9f2a63ff8004910a6a0ec7ff65751146cee68)

booo, 2016-05-12:
https://github.com/freifunk-berlin/firmware/blob/master/packages/default.txt#L54 I guess it's a dependency of this.

ffcb, 2016-05-12:
i have tested at my nexx wt3020:

`Installing luci-i18n-splash-de (git-16.132.22399-7a4741e-1) to root...
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips/236/packages/luci/luci-i18n-splash-de_git-16.132.22399-7a4741e-1_all.ipk.
Installing luci-app-splash (git-16.132.22399-7a4741e-1) to root...
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips/236/packages/luci/luci-app-splash_git-16.132.22399-7a4741e-1_all.ipk.
Configuring luci-app-splash.
Configuring luci-i18n-splash-de.`

installing the language-file includes the package of the splashpage

closed 2016-05-12 by SvenRoederer