freifunk-berlin/firmware #356: tc in Kathleen?

smilebef, 2016-05-19:
Die weimar-Firmware hält einen patch tc-patch zur Unterstützung für olsrd.
Ist der auch in Kathleen?

SvenRoederer, 2016-05-20:
wir nntzen den OLSRd, wir er aus dem openwrt-routing-feed kommt (master)

sarumpaet, 2016-05-21:
@smilebef Bitte aussagekräftigen Betreff, Beschreibung und Link, so bringt das doch nichts.

glycoknob, 2016-05-21:
Hi, Martin vom Weimarnetz hier - glaub es geht um das hier: https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/init.d/S90olsr-prio 

Original ist das wohl aus der Sven-Ola Firmware und es wird die prio qdisc genutzt um oslr Pakete in der Schlage nach vorne zu schubsen. 

Das Problem war wohl, dass wenn man viel Traffic über einen Node hat der über Wifi mehre Nodes traversieren muss nach einiger Zeit ETX so schlecht wird, dass olsr teilweise die Routen verwirft oder neue Routen wählt und damit temporär das Netz lahmlegt. 

Soweit die Theorie - wir haben das nie wirklich getestet, es soll wohl ein wenig helfen aber die prio qdisc verwirft keine Pakete sondern sortiert nur die Warteschlange um? 

Eine Mögliche Alternative wäre noch WMM (QoS im WLAN-Treiber) um olsr Pakete mit höhrerer Priorität rauszuwerfen. 

Also wäre cool, wenn ihr das mal testet, aber ich bin mir absolut nicht sicher ob es was bringt.

sarumpaet, 2016-05-21:
Ahja, davon habe ich schonmal gehört, soll flatternde Routen bei hohem Traffic verhindern, ja. Ich lasse das Issue mal offen für später, danke.

glycoknob, 2016-05-21:
@sarumpaet genau - das hilft aber nur teilweise, weil es in der Form nur ausgehenden Traffic kontrollieren kann - vielleicht ist mit WMM und der neuen cake qdisc noch was rauszuholen... aber bei viel Verkehr in der Luft ändert sich ja auch die Signalstärke und ähnliches. olsr2 macht das wohl überflüssig.