freifunk-berlin/firmware #357: qos/generate.sh all

bobster-galore, 2016-05-21:
Twimc,

in kathleen 0.1.2 
/usr/lib/qos/generate.sh all 
produced something like
<code>
insmod cls_u32 >&- 2>&-
insmod em_u32 >&- 2>&-
insmod act_connmark >&- 2>&-
insmod act_mirred >&- 2>&-
insmod sch_ingress >&- 2>&-
insmod cls_fw >&- 2>&-
insmod sch_hfsc >&- 2>&-
insmod sch_fq_codel >&- 2>&-
ifconfig br-wan up txqueuelen 5 >&- 2>&-
tc qdisc del dev br-wan root >&- 2>&-
etc.
</code>

now (kathleen 0.2.x) there is only
<code>
/usr/lib/qos/generate.sh: export: line 1: CONFIG_'Default'_enabled: bad variable name
</code>
on wr1043V1 and cpe210V1.1.

Can somebody prove this on same or different routers?
Found it when trying to document tc.

Best regards
bob

bobster-galore, 2016-05-26:
Well, well, since we've got a request from the man out in the lorry, we investigated further.

Unfortunately the script does not only do the marking stuff of iptables but also creates all the traffic control (tc) thingies. When it stops at line 1 nothing is done, so there is no more tc. What will lead to link congestion when uploading a lot (download equals upload bandwidth). 

I can't understand line 1 of /usr/lib/qos/generate.sh fluently nor get the kick of /lib/functions.sh, so it would be nice if sb talking script would take over?!

Is Kaya running alpha on his 5GHz-stations, he told me yestern8 he experienced funny loads at one station, could be related?

cheerio bob

SvenRoederer, 2016-05-28:
build "kathleen 0.2.0-alpha+4d87f79" (3. Apr 2016) is also affected.
So it's also before the firmware-hacking event.

bobster-galore, 2016-06-04:
Yestern8 I tested my cpe again, it works as expected.
The only difference is, that it went to a church for measuring and back to my window. I don't believe that it's related to goddebug so its may be because of cold boot?
Could pls someone crosscheck cpe210 behaviour after dis-/reconnecting power!
If it's the reason probably a hint to cold boot would be useful?!

bobster-galore, 2016-07-01:
Probably the whole tc stuff is sometimes working and sometimes not working due to the installation / configuration situation. Depending on cabling during configuration interfaces are up or not and scripts which only run once will not do the necessary preparation of the system. 
Changing switch configuration will evoke scripts for qos?
Which scripts do only run once?

Another possibility is that some scripts run to fast and are so missing the result of a previous script?

Perhaps when we solved this we take a look at the mesh traffic where it goes? And why it is not touched bei qos rules.

booo, 2016-07-01:
I did a test on a router directly and it seemed to work via lan and wifi ap with the latest and greatest master build for WR842v3. I will do another test for the mesh setup and report back.

bobster-galore, 2016-07-02:
Attached u will find a diff between two routers (Freifunk Berlin kathleen 0.2.0-alpha+530b9a1) cpe210V1.1 and wr1043 V1. Qos is working on cpe not on wr1043.
[diff.txt](https://github.com/freifunk-berlin/firmware/files/344333/diff.txt)

Any hints what to look for? (Why one is working and the other isn't?)

Concerning qos on mesh my thoughts:
- qos relies: 
- a) on physical bandwidth (if is 1GB)
- b) on contract bandwidth (e.g. 16MB)
- c) on usable bandwidth (less than 16MB)

How can this be described for air links?

Will this be a script revoked / invoked everytime a link quality changes?
If the script is started again, all qdiscs which means packets are lost (I think) and communication is started again.

Actually there is no way (I think) to describe new qos-interfaces beside the ones thru the "system" so it might be nice to talk to the originator of the script (ngb or something) to introduce more interfaces.  @open-wrt  there is not a huge focus on mesh (I think).

Thresholds can determine when an interface / qdisc is started again? How u mean?

cheerio b

booo, 2016-07-03:
I can't reproduce your issue with a WR842v3 as gateway running http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/998/VERSION.txt. It works on the local lan ports, on the local wifi and from another node in the network.

The qos script will only generate a proper configuration if the interface e.g. ffvpn is actually up/there. If you run the qos script with out e.g. the ffvpn present you get some errors. Anyway the hotplug qos script will later on generate a proper qos configuration for the interface on the interface up event.

SvenRoederer, 2016-08-05:
@bobster-galore @booo Do you agree to close this? As it seems only to fail when there is no uplink (ffvpn) but w/o uplink no shaping needed

bobster-galore, 2016-08-05:
I agree on closing this issue because "/usr/lib/qos/generate.sh all" works in kathleen 0.2.0-alpha+e4f636a with or without ffvpn.
Perhaps there was later a significant change in code or build process why it's working again?

SvenRoederer, 2016-08-05:
let's keep this in mind and reopen if required.

As far as I remember there wasn't a change ...

closed 2016-08-05 by SvenRoederer