freifunk-berlin/firmware #367: kathleen 0.2.0

lynxis, 2016-06-08:
What's missing for the next release?

SvenRoederer, 2016-06-19:
-  just merge the TPlink CPE LNA-patch
- just added some more GL.Inet devices

SvenRoederer, 2016-06-19:
missing: 
- "detection of Release-version and not including commmit-hash will be done later" from PR #324 --> there is some idea: PR #369 
- unsure situation on QoS / traffic-shaping (see https://github.com/freifunk-berlin/firmware/issues/357) @bobster-galore
- test of https://github.com/freifunk-berlin/buildbot/pull/34
- maybe some buildbot adjustments for "release build"

sarumpaet, 2016-07-11:
* [ffwizard] never try to auto-login if password has been set - https://github.com/freifunk-berlin/firmware-packages/pull/83
* [luci-app-ffwizard-berlin] Remove existing certificate files - https://github.com/freifunk-berlin/firmware-packages/pull/59

SvenRoederer, 2016-07-13:
building a release Version will give "nice" filenames and version-string with branch https://github.com/freifunk-berlin/firmware/tree/unstable_SAm0815_Makefile
- tested with a "0.1.99" release-build
- so we can use this hack for this release-branch only see for a clean solution in master for teh next release

sarumpaet, 2016-07-13:
We don't actually need to list everything here; https://github.com/freifunk-berlin/firmware/milestone/3 is intended for this.

SvenRoederer, 2016-07-30:
I just created branch https://github.com/freifunk-berlin/firmware/tree/0.2.0-beta, by rebasing "master" onto the "unstable_SAm0815_Makefile".
Lets take this branch to work check for proper build and function. Then include PR #373 and rename to "0.2.0" --> release.

in Parallel we can continue to work on master with the open PRs and "uncritical" issues above.

comments welcome ... (@bobster-galore)

SvenRoederer, 2016-08-02:
luci-app-owm was sending "Router-model" & "Chipset-type" to OWM-Server. from here the webpage pulls the data for the statistics.
This is broken by now LuCI-code see #378

sarumpaet, 2016-08-04:
@SvenRoederer Is 0.2.0-beta really master being rebased onto unstable? That's a bit weird - I'd expect it to be the other way around, as we cannot/could not fastforward-merge 0.2.0-beta onto master that way. Also look at https://github.com/freifunk-berlin/firmware/network , it's really starting to look ugly now (0.2.0-beta looks like it was branched from master back beginning of May). Looks like 0.2.0-beta would benefit a lot of `rebase --interactive master` or similar. We should meet for a hacking evening, do the rebase and get the 0.2.0 release (or at least RC) out. :)

lynxis, 2016-08-06:
@SvenRoederer please use the master! I've quite a lot issues with the beta as well with your unstable and don't want to address them here. We have a master, where we discuss changes via pull-request.

SvenRoederer, 2016-08-07:
@sarumpaet this branch came from master some time ago and I did the changes in the Makefile. To not redo them again I rebased the changes from master to it. So this results in the "strage" commit-history, but the pure diff was ok.

As of the current efforts it's maybe better to restart this branch from master and apply the current diff.

SvenRoederer, 2016-08-07:
@lynxis the changes in the Makefile are at the end only https://github.com/freifunk-berlin/firmware/blob/0.2.0-beta/Makefile#L107, https://github.com/freifunk-berlin/firmware/blob/0.2.0-beta/Makefile#L134 and https://github.com/freifunk-berlin/firmware/blob/0.2.0-beta/Makefile#L202.

As we agreed this is not the perfect way we should use and you talked about some other ways of implementing this feature. So I think it's fair to not pollute the master with this hack and have it only for 0.2.0. In parallel we can implement a clean solution in master w/o interfering with this "hack"

lynxis, 2016-08-08:
@SvenRoederer I thought it's already fixed by https://github.com/freifunk-berlin/buildbot/commit/124db464cf116c1584af2a88cf4fbcbb4afc62dc
Let's discuss this topic in a seperate ticket. I guess we can use https://github.com/freifunk-berlin/buildbot/issues/33

SvenRoederer, 2016-08-15:
https://github.com/freifunk-berlin/buildbot/commit/124db464cf116c1584af2a88cf4fbcbb4afc62dc only fixes the use of the correct REPO-path for opkg

I don't think the buildbot is the right end to address the way the resulting filename looks like. This should be done in the Makefile for better portability and standalone-use.

lynxis, 2016-08-19:
@SvenRoederer can you please write down, what's wrong with our naming we have now. and how you would like to see it and please create a new issue for this.

SvenRoederer, 2016-08-20:
@lynxis see #387 for the filename / version-string

sarumpaet, 2016-09-29:
With https://github.com/freifunk-berlin/firmware-packages/pull/83 merged I think we're go? Or is anything else missing? I don't really care about the hash filename things... adding some hack to the Makefile or just renaming on the server is both fine I think.

SvenRoederer, 2016-10-01:
we should quickly decide on https://github.com/freifunk-berlin/firmware/issues/381

and then giving this release out

closed 2016-12-09 by sarumpaet