freifunk-berlin/firmware #371: collectd sometimes doesn't start after reboot

bernardladenthin, 2016-06-19:
Hallo,

mein Router (TP-Link TL-WDR3600) startet per Cronjob neu. Ich habe öfter das Problem, das collectd nicht startet. Gefühltes Vorkommen: 1 von 30 Neustarts.

Derzeit installierte Firmware:
http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/974/default/kathleen-0.2.0-alpha%2Bcb51993-tl-wdr3600-v1-sysupgrade.bin

Das gleiche Problem trat jedoch auch die letzen Monate sehr häufig mit der derzeit stabilen Firmware 0.1.2 auf.

Meine derzeitige Lösung ist, collectd im cronjob öfter blind zu starten.

Bis auf das regelmäßige Neustarten des Routers sind alle Einstellungen default. Welche Möglichkeit habe ich, weitere Informationen zu liefern? In dem System- und Kernelprotokoll habe ich nichts auffälliges finden können.

Vielen Dank und viele Grüße, Bernard

SvenRoederer, 2016-06-19:
ich hatte hier jetzt eine NanostationM2Loco, mit build "kathleen 0.2.0-alpha+4d87f79" (3. Apr 2016), bei der nach ungefähr 2 Monaten plötzlich der collectd "stoppte". 
Logs hab ich auch nicht, da mir das erst eine Woche später auffiel.

Keine Ahnung, ob die beiden Probleme zusammen hängen.

Soilman82, 2016-06-21:
Ich hatte selbiges Problem mit meinem TP-Link TL-WR841ND v1 und der 0.1.2. Der Router startete dort immer mal wieder selbstständig neu und danach lief collectd öfters nicht mehr. Musste erst irgendwas in den Einstellungen im Webinterface ändern und speichern, ab dann lief es wieder. 

Mein Workaround, vor allem um die Abstürze/Neustarts zu verhindern, war dann auf die 0.2.0 build 981 (http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/981/default/kathleen-0.2.0-alpha%2Baca2be0-tl-wr842n-v1-sysupgrade.bin) zu gehen. Ob es jetzt noch zu diesem Fehler mit collectd kommen würde, kann ich leider nicht sagen, da der Router seitdem stabil läuft.

SvenRoederer, 2016-06-25:
o.g. NanoStation M2 Loco hatte nach 5 Tagen mit frischer Firmware (master) auch wieder einen abgestürtzten collectd.

Ich werd auf dem Node mal die "Syslog-buffersize" vergrössern, dass evtl. auch noch nach längerer Zeit was im Log steht. @bernardladenthin @Soilman82 - könnt ihr das auch machen?

zusätzlich ist die Umstellung auf den procd (https://github.com/openwrt/packages/commit/a62ff02b8dec94a8153b42677c3795dcb93a5e7b) hilfreich, wenn auch keine Klärung der Ursache für die Abstürze

bernardladenthin, 2016-06-26:
"Größe des Systemprotokoll-Puffers" hat nun 2048. Okay?
Wo finde ich die Datei?
find -iname collectd.init
liefert mir keine Ergebnisse...

SvenRoederer, 2016-06-26:
du kannst diesen buffer dann mit "logread" auslesen und entsprechend weiterbearbeiten

bernardladenthin, 2017-01-03:
Das Problem tritt bei Kathleen 0.2.0 immer noch auf.

SvenRoederer, 2017-01-04:
@bernardladenthin wie letztens geschrieben: du kannst diesen buffer dann mit "logread" auslesen und entsprechend weiterbearbeiten