freifunk-berlin/firmware #375: Set basic_rate for AP/ad-hoc

sarumpaet, 2016-07-13:
Do not allow clients with very bad signal to throttle down the whole AP. See
https://lists.berlin.freifunk.net/pipermail/berlin/2016-July/033682.html

SvenRoederer, 2016-07-13:
really for 0.2.0 ?
I prefer to do some real-life-test to find a proper setting.

sarumpaet, 2016-07-15:
https://github.com/freifunk-gluon/gluon/pull/467
https://github.com/freifunk-gluon/gluon/pull/810

lynxis, 2016-08-09:
I'm using those for years. I completely agree to deactivate B rates.

SvenRoederer, 2017-04-09:
please see upstream discussion 
https://github.com/lede-project/source/pull/945

SvenRoederer, 2017-05-30:
the wifi-option "legacy-rates" has been added to upstream LEDE to support this (https://github.com/lede-project/source/commit/e194e1b3c838a301178effb639804c28fe67354d)

bobster-galore, 2017-10-07:
Would the suggested solution in https://github.com/freifunk-berlin/firmware-packages/pull/106 be ok with you?

SvenRoederer, 2017-12-25:
PR https://github.com/freifunk-berlin/firmware/pull/499 seems to me like an easy and clean solution

SvenRoederer, 2018-03-16:
b-rates have been disabled by default in Hedy-1.0.0 (https://github.com/freifunk-berlin/firmware/commit/d3c50b6d6710ba09e212fc13c19df74250686b2c).
With a working migration we should do this also for Hedy-1.0.1

closed 2018-03-16 by SvenRoederer