freifunk-berlin/firmware #376: Keine Internetverbindung nach 6 Uhr morgens

bernardladenthin, 2016-07-15:
Hallo,

mein Router (TP-Link TL-WDR3600) hatte kurz nach 6 Uhr kein Internet mehr. Es ist ab 6 Uhr aufgetreten und hat sich nicht selbst behoben. Erst ein Neustart um 10 Uhr brachte Besserung. Vermutlich aus dem Cron (ifup wan).

Derzeit installierte Firmware:
http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/974/default/kathleen-0.2.0-alpha%2Bcb51993-tl-wdr3600-v1-sysupgrade.bin

Vielen Dank und viele Grüße, Bernard

Auszug aus dem Log:
```
Fri Jul 15 05:45:00 2016 cron.info crond[1895]: USER root pid 6819 cmd killall -HUP dnsmasq
Fri Jul 15 05:45:00 2016 cron.info crond[1895]: USER root pid 6820 cmd /usr/sbin/ff_olsr_watchdog
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq[1555]: read /etc/hosts - 1 addresses
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/dhcp - 2 addresses
Fri Jul 15 05:45:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
Fri Jul 15 05:45:05 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:45:05 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:46:38 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:46:38 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:47:30 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:47:30 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:49:02 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:49:02 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:49:57 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:49:57 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:50:00 2016 cron.info crond[1895]: USER root pid 6844 cmd killall -HUP dnsmasq
Fri Jul 15 05:50:00 2016 cron.info crond[1895]: USER root pid 6845 cmd /usr/sbin/ff_olsr_watchdog
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq[1555]: read /etc/hosts - 1 addresses
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/dhcp - 2 addresses
Fri Jul 15 05:50:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
Fri Jul 15 05:51:26 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:51:26 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:52:24 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:52:24 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:53:52 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:53:52 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:54:51 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:54:51 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:55:00 2016 cron.info crond[1895]: USER root pid 6870 cmd killall -HUP dnsmasq
Fri Jul 15 05:55:00 2016 cron.info crond[1895]: USER root pid 6871 cmd /usr/sbin/ff_olsr_watchdog
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq[1555]: read /etc/hosts - 1 addresses
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/dhcp - 2 addresses
Fri Jul 15 05:55:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
Fri Jul 15 05:56:16 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:56:16 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:57:18 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:57:18 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 05:58:41 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 05:58:41 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 05:59:44 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 05:59:44 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 06:00:00 2016 cron.info crond[1895]: USER root pid 6895 cmd killall -HUP dnsmasq
Fri Jul 15 06:00:00 2016 cron.info crond[1895]: USER root pid 6896 cmd /usr/sbin/ff_olsr_watchdog
Fri Jul 15 06:00:00 2016 cron.info crond[1895]: USER root pid 6897 cmd ifup wan
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq[1555]: read /etc/hosts - 1 addresses
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/dhcp - 2 addresses
Fri Jul 15 06:00:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
[... 185 lines cut ...]
Fri Jul 15 06:25:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 06:25:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
Fri Jul 15 06:25:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/dhcp - 2 addresses
Fri Jul 15 06:25:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
Fri Jul 15 06:25:19 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 06:25:19 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 06:26:35 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 06:26:35 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 06:27:43 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 06:27:43 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 06:29:01 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 06:29:01 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
Fri Jul 15 06:30:00 2016 cron.info crond[1895]: USER root pid 7750 cmd killall -HUP dnsmasq
Fri Jul 15 06:30:00 2016 cron.info crond[1895]: USER root pid 7751 cmd /usr/sbin/ff_olsr_watchdog
Fri Jul 15 06:30:00 2016 daemon.info dnsmasq[1555]: read /etc/hosts - 1 addresses
Fri Jul 15 06:30:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/odhcpd - 0 addresses
Fri Jul 15 06:30:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr.ipv6 - 7 addresses
Fri Jul 15 06:30:00 2016 daemon.info dnsmasq[1555]: read /tmp/hosts/olsr - 7 addresses
[... cut ...]
Fri Jul 15 06:30:00 2016 daemon.info dnsmasq-dhcp[1555]: read /etc/ethers - 0 addresses
Fri Jul 15 06:30:07 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 
Fri Jul 15 06:30:07 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.110 5c:0a:5b:51:67:10 android-564e75ed3af35f1f
Fri Jul 15 06:31:28 2016 daemon.info dnsmasq-dhcp[1555]: DHCPREQUEST(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee 
Fri Jul 15 06:31:28 2016 daemon.info dnsmasq-dhcp[1555]: DHCPACK(br-dhcp) 10.230.121.105 f4:09:d8:d5:97:ee android-3f4c7c945f0487aa
```

booo, 2016-07-15:
Nach einem Ausfall der VPN-Server Hardware sind die VM-Instanzen für das VPN03 nicht richtig gestartet und haben teilweise keinen Traffic ins Internet geroutet. Die Probleme sollten jetzt behoben sein.

Danke für den Hinweis!

closed 2016-07-15 by booo

bernardladenthin, 2016-07-15:
Vielen Dank!