freifunk-berlin/firmware #378: OWM stats broken (router type/chipset)

sarumpaet, 2016-07-31:
...see http://berlin.freifunk.net/network/stats/ - owm.lua currently doesn't send info on the router type and chipset (was commented out when moving to CC since the luci API changed), which makes statistics useless. We should fix this before releasing 0.2.0 if we want to keep an overview of what hardware is actually used in the mesh.

SvenRoederer, 2016-07-31:
Are you referencing to the "unknown" type?

sarumpaet, 2016-07-31:
Yes. See https://github.com/freifunk-berlin/firmware-packages/commit/e62afa705ae8d145b7e483d2a394752636f62e6d#diff-1190ed871db168151a64d3f9bd15b0ceL241

ubus system/board might give some of the info needed: https://git.lede-project.org/?p=project/luci.git;a=blob;f=modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm;h=8c3e01e603ba79a4d58025c39e43eefc8ae45231;hb=1367f7b624e9ee926cdbd3235a9fec4450c904b5#l655

SvenRoederer, 2016-08-01:
"ubus call system board" --> system is a way to go
"cat /proc/cpuinfo" --> system type gives the same

this seems a way to go. probably the ubus way is more protable / consistent than cpuinfo.

SvenRoederer, 2016-08-01:
Is the result of
> ubus system/board might give some of the info needed

the same as in the previous release?

changed meaning / different expression for the same board will not make the stats-page really usefull

bobster-galore, 2016-08-01:
Hej you,

I took a look at the system-field in json files, their content is slightly different.

json-file Kathleen 1.2
> "system":{"sysinfo":["Atheros AR9344 rev 2","TP-Link TL-WDR3600 v1\n",126652,12732,3720,82280,278.93000000000000682,0,0,0],"uptime":[1982199],"loadavg":[0.03369140625,0.09814453125,0.126953125]}

json-file Kathleen 0.2 alpha
> "system":{"sysinfo":[{"memory":{"buffered":3633152,"total":63037440,"shared":1077248,"free":25792512},"uptime":1480076,"localtime":1470083340,"load":[3008,6912,8512],"swap":{"free":0,"total":0}}],"uptime":[1480076],"loadavg":[0.0458984375]}

Ideas?

SvenRoederer, 2016-08-02:
I manually added the ubus (system->boardinfo->system) to the hardware field on my WR-1043v2 (Ahof-frieden03.olsr)  on 1. Aug. 2016 afternoon. 

> ... ,"hostname":"Ahof-frieden03","hardware":"Qualcomm Atheros QCA9558 ver 1 rev 0 - SAm0815","latitude":52.436757800213996461,"updateInterval":3600, ...

But this still is not showing up in the statistics page. The "hardware" field is updated on the OWM-database and can be read also via "full JSON datasheet" on the Berlin-Map of the node

SvenRoederer, 2016-08-02:
@bobster-galore not sure about the syntax-change you found.

But the "sysinfo"-field may be the relavant for chipset-type, as the "hardware"-field seems to have no effect

sarumpaet, 2016-08-02:
@SvenRoederer Ahof-frieden03 didn't show up because the OWM bounding box was set too small for the stats page. Fixed in https://github.com/freifunk-berlin/berlin.freifunk.net/commit/d5f009c77308a4b90a5387254f0811c31218e5ea . We have 744 nodes now as a side effect. :)

What about the router (hardware) name?

SvenRoederer, 2016-08-02:
@sarumpaet the chipset-thing (hardware) is showing up now

sarumpaet, 2016-08-02:
Yes. We'll also want `ubus system->board->model` stored in `(owm-json).sysinfo[1]` or, probably better, added as `(owm-json).model` (and adjust https://github.com/freifunk-berlin/berlin.freifunk.net/blob/master/www/network/stats.html#L113 accordingly).

SvenRoederer, 2016-08-02:
so switching to "(owm-json).model" would
* break with all old releases
* require checking old and new json-fields in statistics-page

I don't like the idea of checking two fields. But I expect that it's more unrealistic to get all nodes updates to the new code in a reasonable time

bobster-galore, 2016-08-02:
@SvenRoederer will sysinfo not be used?

SvenRoederer, 2016-08-02:
@bobster-galore in current LUCI-API luci.sys.sysinfo() is not avail anymore

SvenRoederer, 2016-08-02:
I suggest the following:
* To get the OWM-upload working let's backport the relevant code from old LuCI (https://github.com/openwrt/luci/blob/luci-0.11/libs/sys/luasrc/sys.lua#L170) - variable: system & model. Let's do this only for the 0.2.0 branch
* Let's think of a way for a clean implementation of new field `(owm-json).model` and other fields (`uptime`, `loadavg`, `mem`) as started in the 0.2.0-alpha-code referenced by @bobster-galore for the next release

SvenRoederer, 2016-08-03:
I have updates the code ("ported the sysinf() from ols luci), to make it work with kathleen 0.2.0.
confirmed to pop up in stats (1 x "TP-Link TL-WR1043N/ND v2" (ahof-frieden03.olsr); 1 x "TP-Link TL-WR1043N/ND v1" (ahof-frieden01.olsr), 1 x "TP-Link TL-WR842N/ND v3"). But for some reason the WR1043-types are not summed up to the other (from 0.1.2-code) WR1043-models.

bobster-galore, 2016-08-03:
probably because of the end "TP-Link TL-WDR3600 v1\n"?

SvenRoederer, 2016-08-03:
yepp, it's the tailing "\n"

bobster-galore, 2016-08-04:
> "system":{"sysinfo":["depreciated value system","TP-Link CPE210 v1.1\n"],"uptime":[1721276],"loadavg":[0.0390625]},
if this is the desired output, it's working!
The deprecated field is neglected because chipset comes from "hardware"?

SvenRoederer, 2016-08-04:
OWM-stats only checking the "Routertype field" in sysinfo and the "hardware field".
So I like to "announce", that this sysinfo-array-datafield will become obsolete and will change.

If there are no concerns I'll open a new issue where we can think of reworking the sysinfo-field and in parallel I'll merge this "hack" to keep compatibility to to current OWM-stats-code

sarumpaet, 2016-08-04:
Yes, that's what I meant in https://github.com/freifunk-berlin/firmware/issues/378#issuecomment-236907904 .

bobster-galore, 2016-08-04:
May I ask where we collect the ideas for the next (past 0.2.0) release?

SvenRoederer, 2016-08-15:
merged in https://github.com/freifunk-berlin/firmware-packages/commit/ac64aa1650588fd1ce96bbf949b0d47c1ab4a57e

closed 2016-08-15 by SvenRoederer