freifunk-berlin/firmware #380: ff_olsr_watchdog ignoriert olsrd disable

SvenSFS, 2016-07-31:
Hallo,
mir ist aufgefallen das der ff_olsr_watchdog den Umstand
ignoriert das olsrd bei mir abgeschaltet ist.

Aufgrund der massiven last die olsrd verursacht hab
ich den olsrd abgeschaltet und über /etc/init.d/olsrd disable
gesetzt.

Jedoch wird das ignoriert, ich musste das über dir crontab 
auf die harte Tour abschalten, also mit # aus dokumentiert.

Sven

SvenRoederer, 2016-08-01:
"ff_olsr_watchdog" gibt's im nächsten release (kathleen 0.2.0) nicht mehr

jal2, 2016-11-02:
@SvenSFS 
Gibt es bei nach dem Abschalten mit "/etc/init.d/olsrd disable" und einem Neustart die Files /var/run/olsrd*.pid denn noch? Eigentlich arbeitet ff_olsr_watchdog nur dann, wenn es olsrd.pid oder olsrd4.pid dort vorfindet. Was bei Dir nicht der Fall sein sollte. Oder ich verstehe den Lua Code nicht.

booo, 2016-11-03:
Alternativ an die Interfaces in der olsrd-Konfiguration "option ignore 1" schreiben und der OLSRD sollte nicht mehr dauerhaft gestartet werden.

sarumpaet, 2016-11-07:
ff_olsr_watchdog ist noch da, nur der ffwatchd ist weggefallen. Aufgerufen wird ff_olsr_watchdog in /etc/init.d/freifunk. Siehe auch https://github.com/freifunk-berlin/firmware/issues/400 .

booo, 2017-01-24:
Nach einem `/etc/init.d/olsrd disable` und dem Entfernen der Pid-Files, des olsrd.watchdog Files und der generierten Konfiguration im /var/* startet bei mir der OLSR nicht mehr.

Kann ff_olsr_watchdog kein Pid-File finden oder sollte /var/run/olsrd.watchdog nicht vorhanden sein, wird auch kein OLSR-Prozess gestartet.

Ich denke der ff_olsr_watchdog funktioniert so wie es von den Erfinderinnen intendiert war, deshalb schließe ich das Ticket hiermit. Ob das ein sinnvoller Watchdog ist, sollten wir ggf. in einem anderen Ticket diskutieren.

closed 2017-01-24 by booo