freifunk-berlin/firmware #381: freifunk-assistent:  too high tx-power on NSM2 (maybe others too)

SvenRoederer, 2016-08-04:
We just flashed a NanoStation M2 with "kathleen 0.2.0-alpha+25c5e44" and TX-power is set to 30dBm. This seems also affect old kathleen 0.2.0-alpha+e4f636a.

lynxis, 2016-08-06:
I can not find any reference in the package repository nor in this firmware repository about txpower. Can you please verify it?

SvenRoederer, 2016-08-07:
I checked again on a factorynew Nanostation M2 (AirOs 5.6.2) flashed with build "25c5e44".
Just started the Freifunk-assistant and ran through (no VPN-Setup) - after reboot the TX-power was @ 30dBm.
As the Wifi is setup by the assistant, I assume the cause is there or at least related.

SvenRoederer, 2016-08-07:
Just checked after defaulting this unit again: The default Wifi config (from OpenWRT) sets the tx-power to max. 
Even after changing the Reg-domain to DE, removing the current Wifi-AP and creating a new, the TX-Power is @ 30 dBm.
So it seems to be related to upstream OpenWRT.

SvenRoederer, 2016-08-07:
maybe helpful:
after setting the RegDomain to "DE" and rebooting the setup still looks strage:
>root@OpenWrt:/# iw reg get
global
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
>
phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)
>
global
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

The phy0 seems not to honor the RegDB-setting

lynxis, 2016-08-09:
do you rebooted after setting the regdb to DE?

bobster-galore, 2016-08-09:
What do you think of querying the owm-API for txpower > 20?
I can imagine that we get a fast overview how many routers are affected and if it is only NSM2. Later we can send information about high txpower on hostname to nickname and mail.

Harald wrote on ff-ml "that default exists from Kathleen 0.0.0 since the nanostations at Christophorus-Kirche were also affected".

The shortest way for kath 0.2.0 would be autocorrection thru the wizard or?

lynxis, 2016-08-10:
@SvenRoederer which value do you read to know it's using 30dbm txpower?

SvenRoederer, 2016-08-10:
@lynxis - sure even after reboot this iw-output

lynxis, 2016-08-10:
I compared a nano-xw/5ghz between lede & oem firmware using register dump of beacon power.
lede iw wlan0 info - txpower 17dbm == 22dbm within the webinterface of the firmware.
IMHO: the webinterface should show the value including antenna. so it's not so far off as you sad. But I don't even trust the values of the vendor.
The iw reg get showing still the US values is a bug.

lynxis, 2016-08-10:
the antenna gain should be already included within the calucaltion of the txpower. it's part of the calibration table.

SvenRoederer, 2016-08-10:
right, the antenna-gain seems to be know by Openwrt. This is the value "TX power offset" shown in iwinfo, which I referenced as `with 12 dBi of antenna`.The TXpower is set to 18 in UCI, which sums up to 30dBm. this 30dBm are finally shown by iwinfo and also in the LUCI-webpage.
The `iw`-tool doesn't know about antennas and only prints the pure RF-txpower (18 dBm), w/o any knowledge of the attached antenna.

SvenRoederer, 2016-08-10:
and of course, the 18dBm (for iw) are within the RegDom-limit of 20dBm. The kernel and mac80211-layer don't know of antenna-gains.

So it seems to be an `iwinfo`-problem, not honoring the whole tx-chain and checking for correctness. 

@lynxis regarding your reading of beacon-power: 
* is this measured on the transmitting device? 
* the seen difference from 17 dBm to 22 dBm may be related to calibration or something like this. And all this is more estimating than measuring, but good to find a rough value
* If your measurement was done on the TXing device and was roughly around 20 dBm in the PHY, I believe there is no knowledge of the antenna-gain and when you add this (11 dBi on NSM2) you are far off the limit for ETSI 2G4Hz

lynxis, 2016-08-10:
@SvenRoederer
* is this measured on the transmitting device?
 * Yes. I read the beacon power out of the register 0x181080E8 with the oem firmware, as well as with LEDE.
* the seen difference from 17 dBm to 22 dBm may be related to calibration or something like this.
 * i don't think so. but there might be txgain tables and others. But I guess it's the same. Only real hardware measurement as elektra successted could help here.
* If your measurement was done on the TXing device and was roughly around 20 dBm in the PHY, I believe there is no knowledge of the antenna-gain and when you add this (11 dBi on NSM2) you are far off the limit for ETSI 2G4Hz
 * The antenna gain is part of the calibration. So yes, the driver knows about the antenna gain.

SvenRoederer, 2016-08-18:
I did some test with a NSM2 with original AirOS and OpenWRT as described in https://lists.berlin.freifunk.net/pipermail/berlin/2016-August/033935.html .
* Screenshot from Android-phone with AirOS: https://lists.berlin.freifunk.net/pipermail/berlin/attachments/20160816/337f2909/attachment.png (see WiFi "ubnt")
* Screenshot from Android-phone with OpenWRT: 
https://lists.berlin.freifunk.net/pipermail/berlin/attachments/20160816/337f2909/attachment-0001.png (see WiFi "OpenWrt")

In the setup the reference-WiFi "WR741" is quite set same level, but there is a very big difference in the RF-Levels. Both stations are setup to default level (OpenWRT: 30dB in LUCi; AirOS: 9dB in Web-UI)

SvenRoederer, 2016-09-07:
see https://github.com/freifunk-gluon/gluon/issues/94 where this issue also came up

bobster-galore, 2016-09-29:
Would it be a solution to reduce the tx-power of concerned devices to a legal rate using a hw- / tx-power-table? 
Then there would be no need to ask users to check their settings. To cite offline-Heinz "quiet is the new loud." In cases where more power is needed, users can increase the power.
On Teufelsberg we found nsms shouting from Christophorus app. 5km away until we reduced tx-power to reasonable values.

SvenRoederer, 2016-09-29:
The question is: Are we willing to do something, as we have clear indications that there is to much TX-power. To ignore this "knowledge" will not help anyway.

Implementing a quick lookuptable for certain board to set new tx-power by the assistent seems easy.

sarumpaet, 2016-09-30:
I don't think the assistent is the right place for that; it's not present in the backbone image, for example. We could add a `if device=NSM2 then uci set wireless.radio0.txpower=8&&uci commit&&wifi` line or similar to migration though (possibly before the guard so that it's run always on upgrades).

bobster-galore, 2016-09-30:
Is it time to start that table?
Perhaps it's better to check the value first like
`if device=NSM2 and txpower>"legal value" then uci set wireless.radio0.txpower="legal value"&&uci commit&&wifi`
So it will not change correct values or manual set values in legal range.

SvenRoederer, 2016-09-30:
@sarumpaet the idea to have it in the assistent is the assumption, that ppl who use the assistent are not so advanced. They rely on the assistent to give them a well-configured router.
Ppl. using the backbone image are more advanced and should / have to take care for their settings (assistent not included)

@bobster-galore when this setting is changed in the assistent it will only change on routers that are reconfigured / initial-configured

bobster-galore, 2016-10-01:
Sven and me propose to switch to legal values by a startup runonce script for all since we found that even experienced users may forget to use correct settings. The script will look up the board name compare set  tx-power to legal tx-power and reduce if necessary.

sarumpaet, 2016-10-02:
I propose to just make it a new config default (e.g., add a file `/etc/uci-defaults/freifunk-berlin-hardware-defaults` with `guard "nsm2_txgain"`), then the setting will be applied both to new installations and upgrades (once). Please don't do any comparison; we don't do this on other routers, so why here. If people really want to set the power higher, why not; there are a few legitimate cases for this (e.g., manually modified antenna hardware).

SvenRoederer, 2016-10-02:
I agree in guard-ing this.
But I don't see a reason for setting the default unconditionally. When the value was set to a lower setting than the new default, the user must have changed it for a reason. Why sould we force him to the higher value he has to lower then again.

sarumpaet, 2016-10-02:
Ah ok, I guess I misunderstood the "startup runonce" part in bobster-galore's comment. Comparing once on migration is good (not on each startup).

SvenRoederer, 2016-10-03:
As I don't find a way to distinguish between NSM2 and NSM2loco, I suggest to use an average of the antenna gain. 
* NSM2: up to 11.2 dBi
* NSM2loco: 9 dBi
* avg: 10dBi (NSM2loco will be a bit tp low, NSM2 might be a bit above ...)

Any concerns?

sarumpaet, 2016-10-04:
Fine for me :+1:

lynxis, 2016-10-04:
Why not fixing this issue for real on openwrt/lede devices? This is just a work-around.
As stated, the offset is 5dbm and not 9 or 12.

SvenRoederer, 2016-10-04:
@lynxis Fixing this upstream might take some time and somebody capable to do this. so the Work-around for a known problem seems valid.
As to my test I stick to the values suggested by Gluon (https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#transmit-power-on-ubiquiti-devices).

SvenRoederer, 2016-10-08:
@bobster-galore  @sarumpaet see https://github.com/SvenRoederer/freifunk-berlin-firmware-packages/blob/fix/issue381/utils/freifunk-berlin-network-defaults/uci-defaults/freifunk-berlin-default-txpower

This uses "iwinfo" to lookup its Antenna-gain and subtracts this from the ETSI maxTX-power .
It runs only once per router, which might be problematic when new routermodels are added to iwinfo or the gain-definition changes.
But this might not be a common problem

SvenRoederer, 2016-10-15:
tested this on several Nanostation + WR1043 + WR741 and works.
https://github.com/freifunk-berlin/firmware-packages/pull/88

any concerns?

booo, 2016-10-28:
Only handling the NSM2 is wrong. What about the other devices with 5GHz or from other vendors (TP-CPE210/510) which are in use in the network?

I do think the lookup table of iwinfo is a wrong approach because it does not account for multiple radios plus antennas on a board. Somewhere there should be a description of the board and the attached antennas and from this information the OS should set sane defaults. A possible place for these defaults is maybe [here](https://github.com/openwrt/openwrt/tree/master/target/linux/ar71xx/base-files/etc/board.d).

I actually think we should push a release without a solution and just add a hint to the release note that there might be a problem with the output power.

SvenRoederer, 2016-10-28:
@booo fyi: the TPlink CPEs are not affected, as the board (calibration?) allows not to set more than 11dBm on a CPE210. With the 9dBi of the antenna all is withi limits. I assume the same for the CPE510.

SvenRoederer, 2016-10-28:
What against the iwinfo-table? 
I'm pretty sure it can handle multiple different radios on a board. Would be great to have such board you describe in hands for testing.

As this problem is also known to the Gluon-guys we should check with them on a way to go. OpenWRT seems not interested in power-limiting, as this seems "individual configuration".

SvenRoederer, 2016-10-28:
considering the arguments:
- multiple-radios
- crazy hardware
- iwinfo currently only knows about Ubiquiti-hardware
- just vague infos on other boards

I'll reword the script to only act on NSM2 / NSLocoM2. But I'll keep the lookup if board and tx-offset via iwinfo. iwinfo seems to be the only reliable way to figure out if we run on a Loco / non-Loco board

commit 0e240ed8 (2016-11-03): feeds: update packages-berlin

- fix #208: redirect to wizard
  https://github.com/freifunk-berlin/firmware/issues/208
- fix #381: too high tx-power on NSM2 (maybe others too)
  https://github.com/freifunk-berlin/firmware/issues/381
closed 2016-11-03 by SvenRoederer