freifunk-berlin/firmware #382: OWM stats: not well defined JSON structure

SvenRoederer, 2016-08-05:
in issue #378 we found that the JSON-dataset uploaded by luci-app-owm has a bad structure.
The system->sysinfo-field is just an array of values, w/o fieldnames. So changing the datafields, as required by the LuCI-API change, causes some trouble to stay compatible with current parsers, like http://berlin.freifunk.net/network/stats/ 

json-file Kathleen 1.2
> "system":{"sysinfo":["Atheros AR9344 rev 2","TP-Link TL-WDR3600 v1\n",126652,12732,3720,82280,278.93000000000000682,0,0,0],"uptime":[1982199],"loadavg":[0.03369140625,0.09814453125,0.126953125]}

It should be something like:
> "system":{
   "systeminfo": {"uptime":1982199,"loadavg":[0.03369140625,0.09814453125,0.126953125], "memory":{"buffered":3633152,"total":63037440,"shared":1077248,"free":25792512}},
   "hardwareinfo": {"model":"TP-Link TL-WDR3600 v1", "chipset": "Atheros AR9344 rev 2"}
   }

So the datafields have individual names and can be parsed in a normal way.

Changing this definition will for sure require some thoughts of a sane way of migration, esp. for lazy updaters who stay on old code.

bobster-galore, 2016-08-05:
Could http://berlin.freifunk.net/network/stats/ parse json files
if it finds "systeminfo" due to the new definition
elif it finds "sysinfo" as old style?
So both versions are supported.

Concerning the content of the json dataset I like to ask if all fields are necessary?
E.g. there are fields which are also collected by collectd.
There are fields which have to be calculated to find, what the message is like uptime and memory. Could it be nice to group the digits for easier reading?
What use is in 11 decimal places in "loadavg" which is an average value?!

Are there any programs which rely on the numbers?

sarumpaet, 2016-08-12:
1) Yes, the stats can parse the data as needed, it's just some JavaScript. 2) collectd and OWM are two completely separate things and should be kept separate as not all nodes can/want to run collectd. 3) Number formatting: Who cares, the JSON is not intended for direct presentation (but if you want, create a PR). 4) I'm not aware of any consumers of the OWM data apart from OWM and the website stats.

It has to be noted that OWM has been due for replacement for a long time. http://libremap.net/ is it's "official" successor (using a different API and scripts as far as I know), but there are many other good candidates for Freifunk maps.

SvenRoederer, 2016-08-15:
@sarumpaet so we should not spend to much effort in fixing OWM-things, right?

What about opening an issue requesting the transition to a current mapping-project and poll for the "best" successor.

sarumpaet, 2016-09-25:
Perhaps we should change the approach a bit:
* instead of having an "OWM" package on the router, have a "ff-berlin-stats" package (or similarly named) instead
* that package collects some specific info (router type, memory stats/uptime, position, mesh info, contact info) and sends that to a berlin.freifunk.net endpoint. This avoids the current "grab all info that's available or might be interesting and dump it into a repository that aims not to be Berlin specific" approach. Why do that? Because dumping random info is a data privacy problem, and dumping it into a repository/map that is not Berlin specific makes it difficult to integrate with our other infrastructure (wiki, monitor). There's a reason why other communities have their own region-specific map installations.
* the server can then reprocess the data and convert it/make it available as needed, and we could support multiple map frontends.

sarumpaet, 2021-04-12:
Let's continue there: https://github.com/Freifunk-Spalter/packages/issues/159

closed 2021-04-12 by sarumpaet