freifunk-berlin/firmware #386: Auskommentierte lokale Variable 'community' in den alpha- und beta-Versionen

AnnieYousar, 2016-08-10:
Gibt es einen Grund, warum in Zeile 46 von index.htm in den alpha- und beta-Versionen die lokale Variable 'community' nicht gesetzt wird:
--local community = uci:get_first(co, "community", "name") or "Freifunk"
Damit wird der Wert 'berlin' der globale Variablen  'community' übernommen und folglich nicht der Wert 'Freifunk Berlin' aus dem profile_berlin.

SvenRoederer, 2016-08-10:
um welches Software-paket geht es?

AnnieYousar, 2016-08-11:
Ja, das war ein wenig blöd, so ohne konkrete Angabe. Tut mir leid.
Hier zwei Beispiele:
http://buildbot.berlin.freifunk.net/buildbot/stable/0.2.0-beta/ar71xx/default/kathleen-0.2.0-alpha%2B8f61f29-archer-c5-sysupgrade.bin
http://buildbot.berlin.freifunk.net/buildbot/unstable/mpc85xx/914/default/kathleen-0.2.0-beta%2B55c6caf-tl-wdr4900-v1-sysupgrade.bin
/Ann.

SvenRoederer, 2016-08-15:
@AnnieYousar der gennutzte Firmware-build (8f61f29 bzw. 55c6caf) ist ein guter Hinweis. 
Aber es ist immer noch nicht klar, wo genau dein gefundenes Problem liegt. Die Firmware besteht ja doch aus einer Menge Code und diversen Dateien. In welcher Datei hast du denn den genannten Code gefunden? Damit könne wir da erstmal rausbekommen, welches Paket diesen Code beisteuert.

AnnieYousar, 2016-09-15:
Jetzt hab ich 's gefunden: luci-mod-freifunk-ui_0.0.1-2_ar71xx.ipk
und da dann /usr/lib/lua/luci/view/freifunk/index.htm
```
--- usr/lib/lua/luci/view/freifunk/index.htm    2016-08-15 14:29:16.000000000 +0200
+++ usr/lib/lua/luci/view/freifunk/index.htm.new        2016-09-15 08:56:38.208600000 +0200
@@ -44,5 +44,5 @@

 local co = "profile_" .. community
---local community = uci:get_first(co, "community", "name") or "Freifunk"
+local community = uci:get_first(co, "community", "name") or "Freifunk"
 local url = uci:get_first(co, "community", "homepage") or "http://www.freifunk.net"

```

AnnieYousar, 2016-09-15:
Die eigentliche Frage war, welche Profildaten, genauer gesagt, welcher Community-Name nun der "richtige" ist, der aus
`uci show freifunk.community`
oder aus
`uci show profile_berlin.profile`.

SvenRoederer, 2016-12-30:
Ich gehe mal davon aus, dass das Problem in Kathleen-0.2.0 auch noch aktuell ist, richtig?

AnnieYousar, 2017-01-04:
Ja. Die "Willkommenseite" sieht damit etwas dürftig aus. Vielleicht ist es zweckmäßig, wenn man eine eigene Freifunk-Berlin-Willkommensseite erstellt.
Etwa so etwas als defaulttext:
if DefaultText ~= "disabled" then
        defaulttext = '<h2><a id="content" name="content">'..
        ("Hallo und willkommen im Netzwerk von")..
        ' '..
        community..
        '!</a></h2><p>'..
        ("Freifunk steht f&uuml;r freie Kommunikation in digitalen Datennetzen. ")..
        ("Wir verstehen <i>frei</i/> als &ouml;ffentlich zug&auml;nglich, nicht kommerziell, ")..
        ("im Besitz der Gemeinschaft und unzensiert.")..
        '<br />'..
        ("Dies ist der Zugangsknoten")..
        ' <a id="content" name="content">'..
        luci.sys.hostname()..
        '</a>. '..
        ("Er wird betrieben durch")..
        ' <a href="'..
        luci.dispatcher.build_url("freifunk", "contact")..
        '">'..
        (nickname or translate("Please set your contact information"))..
        '</a>.</p><p>'..
        ("Weitere Informationen &uuml;ber die globale Freifunk-Initiative findest du unter")..
        ' <a href=\"http://freifunk.net\">http://freifunk.net</a>.'..
        '<br />'..
        ("Wenn du mehr &uuml;ber unser Projekt erfahren willst, dann wende dich an ")..
        ("die lokale Gemeinschaft von")..
        ' <a href="'..url..'">'..community..'</a> '..
        ("oder den Betreiber dieses Knotens.")..
        '</p>'..
        '<p><strong>'..("Notiz")..'</strong>: '..
        ("Der Internetzugang &uuml;ber das experimentelle Freifunknetz ist an technische und ")..
        ("organisatorische Bedingungen gekn&uuml;pft und deshalb m&ouml;glicherweise nicht immer ")..
        ("gew&auml;hrleistet.")..
        '</p>'..
        '<p><strong>'..("Wichtiger Hinweis")..'</strong>: '..
        ("Wenn du dieses Netz nutzt, erkl&auml;rst du damit implizit, dass du im Freifunknetz ")..
        ("keine Rechtsverletzungen begehen wirst.")..'<br />'..
        ("Du stimmst damit auch dem Monitoring durch den Verein ")..
        '<a href="http://freifunkstattangst.de">Freie Netzwerke e.V.</a> '..
        ("und der Kontrolle durch staatliche Organe zu.")..
        '<br />'..
        ("Verlasse das Freifunknetz, wenn du damit nicht einverstanden bist.")..
        '</p>'
end

Aber die Willkommensseite liest ja sowieso keiner ...

AnnieYousar, 2017-01-04:
Ja. Die "Willkommenseite" sieht damit etwas dürftig aus. 
Vielleicht ist es zweckmäßig, wenn man eine eigene Freifunk-Berlin-Willkommensseite erstellt.
Vorschlag in der Anlage.
[welcome.txt](https://github.com/freifunk-berlin/firmware/files/685567/welcome.txt)

SvenRoederer, 2017-01-08:
zur Willkommensseite: Den letzten Satz ("Du stimmst damit auch dem Monitoring durch den Verein <a href=\"http://freifunkstattangst.de\">Freie Netzwerke e.V.</a> und der Kontrolle durch staatliche Organe zu.") finde ich etwas problematisch.

- Das klingt so, als ob der Traffic grundsätzlich gemonitort wird.
- du meinst wohl "Förderverein für freie Netze e.V."
- "Freiheit statt Angst" != Förderverein

SvenRoederer, 2017-01-08:
Zurück zum eigentlichen Problem:
das Paket "freifunk-UI" ist ein Fork vom "luci-mod-freifunk" (https://github.com/openwrt/luci/blob/9780ee382e72f8a5fb69e337a3fcc51fc0914883/modules/luci-mod-freifunk/luasrc/view/freifunk/index.htm#L34). Hier war das auch schon so.

Ich hab mich auch schon mal gefragt, ob es sinnvoll ist, die Daten immer direkt aus dem Community-profil zu lesen, oder in der UCI-Freifunk zwischen zu lagern.
Hier scheint'S generell Unstimmigkeiten zu geben.

AnnieYousar, 2017-01-15:
Schaut man sich die Daten in `uci show | grep profile_ | sed /name/!d` an, dann hat man den Eindruck, als ob mehrheitlich ein `commonName` der lokalen Community verwendet wird (enthält oft Leerzeichen), während es bei `uci show freifunk.community.name` um einen `identifier` geht, mit dem das entsprechende community profile durch `local co = "profile_" .. community`  in `/usr/lib/lua/luci/view/freifunk/index.htm` addressiert wird (also dann auch natürlich ohne Leerzeichen). 
Dem entsprechend finde ich, dass auf der "Willkommensseite" (http://frei.funk) auch der `commonName` erscheinen müsste, der durch 
`local community = uci:get_first(co, "community", "name") or "Freifunk"` aus dem Profil entnommen wird. 
Und bevor du fragst: in dem LEDE-migration Pfad ist es auch noch so.
Ich frag dann mal den Autor der letzten Änderung Jo-Philipp.

SvenRoederer, 2017-11-17:
@AnnieYousar  hast du hier was rausgefunden?

AnnieYousar, 2017-11-21:
Ich habe keine Antwort bekommen. Es bleibt die Frage, welcher Community-Name auf der Willkommenseite verwendet werden sollte:
`uci show freifunk.community.name` (Identifier der Community) oder aus
`uci show profile_berlin.profile.name` (Bezeichnung der Community).
Durch die Auskommentierung wird der erste, also `berlin` und nicht der zweite `Freifunk Berlin` auf die Willkommensseite an die Stelle `community` gesetzt.

closed 2017-11-21 by AnnieYousar