freifunk-berlin/firmware #387: the Commit-hash shoulde not be present  in filename and versionstring of release-builds

SvenRoederer, 2016-08-20:
since PR #324 there is always the git-hash in the filename and the versionstring, this should not be for the release-build.
unstable-builds / current-builds:
> LuCi: Freifunk Berlin kathleen 0.2.0-SAm0815+85b7eea 
shell: Firmware Berlin (kathleen 0.2.0-SAm0815+bab671e
filename: kathleen-0.2.0-SAm0815+85b7eea-tl-wr741nd-v4-sysupgrade.bin

release-builds (as they should look):
> LuCi: Freifunk Berlin kathleen 0.2.0 
shell: Firmware Berlin (kathleen 0.2.0
filename: kathleen-0.2.0-tl-wr741nd-v4-sysupgrade.bin

This is caused by https://github.com/freifunk-berlin/firmware/blob/master/Makefile#L91

SvenRoederer, 2016-08-20:
https://github.com/freifunk-berlin/firmware/compare/0.2.0-beta is a possible way to fix this for the kathleen-0.2.0 release and look for a clean solution for the next release.
Or we merge to master and rework the code from there

lynxis, 2016-08-25:
@SvenRoederer do you tried to `git tag -s` a version and compile afterwards? there shouldn any git hash

SvenRoederer, 2016-08-25:
But to have the git hash in the version-string, of development-builds (master-code), was the intention of PR #324 of @sarumpaet . 
So imho some code to decide between release-build and development-build will be required somewhere.

SvenRoederer, 2018-03-16:
since Kathleen-0.2.0 the only idea / solution for this is this:
* https://github.com/freifunk-berlin/firmware/commit/fe0c25ff606ea2ad0960e4ad7bc09c41f041861c
* https://github.com/freifunk-berlin/buildbot/pull/42

I'd like to merge this into master. It can even replaced by a better solution, as soon there is one.