freifunk-berlin/firmware #388: Lan-Ports eth0 u. eth1 Switch und Schnittstellenkonfigauration

carsifreifunk, 2016-09-14:
Festgestellt auf WR1043v3 und Archer C7 (alles über Weboberfläche)

Die oben genannten Router verfügen über ein eth0 (WAN_Port) und eth1 (LAN-Ports), welche eigentlich keine VLAn benötigen, da eh schon getrennt. Unter Netzwerk -> Switch wird VLAN 1 für Lan auf der CPU als untagged und VLAN 2 für WAN als disabled auf der CPU angezeigt. War an der stelle schon etwas verwirrend. 
Wenn ich ein VLAN 3 hinzufüge muss ich selbstverständlich VLAN 1 und 3 auf der CPU auf tagged setzen, und lasse VLAN 2 disabled.
Wenn ich jetzt eine neue Schnittstelle anlegen möchte habe ich kein eth1.3 zur Auswahl, sondern muss es bei custom selbst eintragen. 
Zusätzlich muss ich noch drauf achten, dass ich meine LAN-Schnittstelle ebenfalls anpasse, indem ich eth1 deaktiviere und bei custom eth1.1 eintrage.

Lässt sich hier noch was fixen?

SvenRoederer, 2016-09-14:
Ich denke, dieses Problem betrifft alle Router, die vom User so umkonfiguriert werden (normaler Lan-port --> Vlan-tagged Port).
Wenn dann sollte man das Upsetram im OpenWRT / LEDE fixen.

P.S: https://wiki.openwrt.org/toh/tp-link/tl-wr1043nd#switch_ports_for_vlans veranschaulicht das Zusammenspiel von LAN-ports, internem Switch und CPU-Ports ganz gut

carsifreifunk, 2016-09-15:
Danke für den Link, der veranschaulicht es wirklich ganz gut.

Aber bei Kathleen 0.1.x war es zumindest noch immer so, dass, wenn ich unter Switch eine VLAN3 angelegt habe, bei den Schnittstellen unter pyhsikalischen Einstellungen dann auch ein eth1.3 zur Auswahl hatte und es nicht manuell eintragen musste. Das gab es nur, wenn kein Switch-Menü zu Verfügung stand.

SvenRoederer, 2016-09-19:
Das kann ich jetzt nicht beurteilen. Aber das ist dann alles OpenWRT-seitig, da werden wir nix ändern :-)

SvenRoederer, 2017-06-11:
there is a Luci-commit (https://github.com/openwrt/luci/commit/426c1043b1464386291df8fe4557ceae81e54166) which reads as it will take care of this issue.

can you check with one of the SAm0815_Hedy-alpha builds (still not pushed to development master)

SvenRoederer, 2017-11-17:
@carsifreifunk had you a chance to check again?

SvenRoederer, 2018-03-16:
AS it should have been fixed upstream and nothing was reported back since  15 Sep 2016 I'll close this one.

Feel free to reopen.

closed 2018-03-16 by SvenRoederer