freifunk-berlin/firmware #391: Change OWM endpoint

sarumpaet, 2016-10-06:
(this is related to https://github.com/freifunk-berlin/firmware/issues/382#issuecomment-249414895)
Map data should be sent to an *.berlin.freifunk.net endpoint instead of owm directly, giving us more flexibility and allowing to switch maps/supporting several maps without hacks.

sarumpaet, 2016-11-08:
See https://github.com/freifunk-berlin/firmware-packages/pull/92

sarumpaet, 2016-11-10:
fixed in https://github.com/freifunk-berlin/firmware/commit/3086c4756d6c31b4d48cd5e8232d1e4b4720be2d and https://github.com/freifunk-berlin/firmware-packages/commit/a6dbc9ffa6e9affb4dc1d40e9ab00077fb33eaa2

closed 2016-11-10 by sarumpaet

SvenRoederer, 2016-11-11:
what's the schedule to have the owm-update function running on the new URL (http://util.berlin.freifunk.net/update_node/)?

sarumpaet, 2016-11-11:
It's running for a few days already (just a quick proxy to openwifimap.net at the moment).

SvenRoederer, 2016-11-11:
As yesterday it was not working and gave no reply. Just checked again and the reply gives "201" and not "nil" anymore.

SvenRoederer, 2016-11-12:
the proxy is still behaving a bit strange:
> root@Ahof-frieden03:~# /usr/sbin/owm.lua 
update Doc  Statuscode: 201 http://util.berlin.freifunk.net/update_node/Ahof-frieden03.olsr (b
doc updated
0
> 
> )
root@Ahof-frieden03:~# 

using original server:
> root@Ahof-frieden03:~# /usr/sbin/owm.lua 
update Doc  Statuscode: 201 http://api.openwifimap.net//update_node/Ahof-frieden03.olsr (doc updated)
root@Ahof-frieden03:~#