freifunk-berlin/firmware #398: Update/fix kernel (CVE-2016-5195)

sarumpaet, 2016-10-24:
* https://en.wikipedia.org/wiki/Dirty_COW
* https://forum.openwrt.org/viewtopic.php?id=68181

SvenRoederer, 2016-10-24:
nice idea to have this, as long as somebody will create a patch we can include.

but as most daemons run a root a remote attacker might gain root-privileges anyway and don't need to use this privilege escalation bug.

sarumpaet, 2016-11-02:
Updating the OpenWrt feed should fix this, see
https://github.com/openwrt/openwrt/commit/2c54d989b46284bc5ee5b43d4ba7c2762c2241ed

SvenRoederer, 2016-11-02:
I don't like the idea to update the whole OpenWRT-base some days before releasing (hopefully).

Beside the small issue of reverting the unification of the imagenames of the TPLink N/ND-devices, there might be other issues come up.

Am 2. November 2016 18:24:30 MEZ, schrieb sarumpaet <notifications@github.com>:
>Updating the OpenWrt feed should fix this, see
>https://github.com/openwrt/openwrt/commit/2c54d989b46284bc5ee5b43d4ba7c2762c2241ed

SvenRoederer, 2016-11-05:
btw. what about CVE-2016-7117 (Use-after-free vulnerability in the __sys_recvmmsg function in net/socket.c in the Linux kernel before 4.5.2 allows remote attackers to execute arbitrary code via vectors involving a recvmmsg system call that is mishandled during error processing.) --> https://github.com/openwrt/openwrt/issues/118

sarumpaet, 2016-11-05:
CVE-2016-7117 might not be fixed in the CC 15.05.1 release but we're using CC HEAD anyways. In Linux 3.18.44 it's fixed.

BTW yay to the people who say we don't need auto-upgrades.

closed 2016-11-12 by SvenRoederer