freifunk-berlin/firmware #400: Reenable watchdog or use procd

sarumpaet, 2016-11-07:
olsrd crashes happen frequently for a number of reasons. We should re-enable ffwatchd or make the switch to procd.

sarumpaet, 2016-11-07:
Actually there are several watchdogs:
* `/usr/sbin/ffwatchd` (from package `freifunk-watchdog`) which monitors the wifi interface and things configured in `/etc/config/freifunk-watchdog` (_not_ olsrd)
* `/usr/sbin/ff_olsr_watchdog` (in package `freifunk-common`) which monitors olsrd1, based on the watchdog timestamp files it keeps current.
I think we only disabled ffwatchd (https://github.com/freifunk-berlin/firmware-packages/blob/1c04e111ceb6be68a9cb83bafe108bacc30abd22/utils/freifunk-berlin-migration/uci-defaults/999_freifunk-berlin-migration.sh#L141 ) so with https://github.com/freifunk-berlin/firmware-packages/issues/90 things might be fine for olsrd. I'm still worried with Dropbear or OpenVPN getting killed on OOM though.

We probably should also fix ff_olsr_watchdog to monitor olsrd6 as well (it doesn't at the moment).

SvenRoederer, 2016-11-07:
As it seems unlikely that the random olsr crashes will be fixed I prefer not to invest energy in porting to procd and stay with watchdog as long a needed.

I favour the idea of migrating to a modern routing-daemon which also supports the IPv6 Sourcerouting-issue preventing from deploying Ipv6 netwide. (see https://github.com/freifunk-berlin/firmware/issues/370 for IPv6-related issues)

booo, 2016-11-08:
`/usr/sbin/ffwatchd` from freifunk-watchdog package should stay disabled.

`/usr/sbin/ffwatchd` (from package freifunk-watchdog) should be configured properly for now. This includes configuration of for the olsrd ipv6 daemon. Maybe we should fix this [upstream](https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-common/files/usr/sbin/ff_olsr_watchdog#L8) and fix our configuration as suggested by @sarumpaet.

Maybe we should collect some information about the OOM-Killer problem in another ticket and investigate the actual cause?

sarumpaet, 2016-11-08:
@booo I guess you mean `ff_olsr_watchdog` in your second sentence.

OOMs can have many causes, and some of them are not really bugs: On 32MB machines, general memory pressure; doing things in the web UI that consume "a lot of" memory (e.g., looking around in the olsr stats); running something "wrong" at the "wrong" time via ssh. The OOM killer isn't very picky and might kill olsr even if olsr is not actually the problem.

We can do something like `echo '-1000' > /proc/$(cat /var/run/olsrd.pid)/oom_score_adj` that prevents the OOM killer from killing olsr but that has other problems: What if, for once, olsr _is_ the actual memory hog?

sarumpaet, 2016-11-08:
For olsrd this should be fixed now; removing the 0.2.0 milestone.

SvenRoederer, 2016-11-08:
about chances that OpenVpn / Dropbear get killed:
there is a risk that this might happen, but will they not be restarted by procd in this case? for Dropbear you might loose you current shell, which can cause problems. But this seems not so common.

SvenRoederer, 2016-11-08:
what about renaming this issue to "Reenable watchdog or use procd for olsr", then we can close this as of merged https://github.com/freifunk-berlin/firmware-packages/pull/90.

And setup a new issue to discuss on the OLSR memory-issue or whatever it is

booo, 2016-11-08:
The `ff_olsr_watchdog` is still broken from my point of view...

booo, 2017-01-24:
By the way there is an uptstream ticket: https://github.com/openwrt/luci/issues/642

Another problem with this ff_olsrd_watchdog: If for some reason the olsrd process does not start once in the life time of the router there is no pid file and no watchdog file. The watchdog decides then not to  restart olsrd. What's the reasoning behind this?

SvenRoederer, 2017-01-24:
it'S a watchdog, to keep the process running or restarting. ff_olsrd_watchdog is not the startup-script which should take care for this.

valentt, 2018-03-16:
Hi,
i just finished writing a detailed blog post regarding how to use hardware watchdog and how to manually take control over from procd:

http://kernelreloaded.com/manually-controlling-openwrt-hardware-watchdog/

Over the years there were few questions regarding this and first there was no way to do it (missing magicclose feature) and documentation was lacking even after it was implemented.

Hopefully this helps some people.

If you have any suggestions or comments regarding this blog post feel free to write here or contact me directly.

Cheers,
Valent.