freifunk-berlin/firmware #401: dnsmasq restart spamming logs

sarumpaet, 2016-11-07:
https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-dhcp-defaults/files/etc/init.d/ff_dnsmasq_reload
@cholin what is/was the purpose of this?

Also, we should consider adding `quiet-dhcp` to `/etc/dnsmasq.conf`.

booo, 2016-11-08:
The SIGHUP triggers dnsmasq to parse the hosts files etc. again. Excerpt from the manpage. 

> "When  it  receives  a SIGHUP, dnsmasq clears its cache and then re-loads /etc/hosts and /etc/ethers and any file given by --dhcp-hostsfile, --dhcp-hostsdir, --dhcp-optsfile, --dhcp-optsdir, --addn-hosts or --hostsdir.  The dhcp lease change script is called for all existing DHCP leases. If --no-poll is set SIGHUP also re-reads /etc/resolv.conf.  SIGHUP does NOT re-read the configuration file."

Maybe the init script should be converted to a firstboot script? I don't get why we start this every time. If we do this we should add some documentation too.

We should add the quiet-dhcp option. It's part of chaos calmer: https://github.com/openwrt/openwrt/blob/chaos_calmer/package/network/services/dnsmasq/files/dnsmasq.init Can you create a pull request or proper ticket for this?

Maybe we should rename this one if we want to work on the "dnsmasq-freifunk" init script?

sarumpaet, 2016-11-08:
see https://github.com/freifunk-berlin/firmware-packages/pull/91

closed 2017-03-23 by SvenRoederer