freifunk-berlin/firmware #405: OLSR looses neighbours asymetrical

SvenRoederer, 2016-11-14:
one of my routers is not having any v4-neighbours, but the neighbours see this router as their neighbour. in IPv6 all is good.

**faulty node: Ahof-frieden03**
 ```
Nachbar-IP 		Hostname 			Schnittstelle 	Lokale Interface-IP
fd2d:21b4:57d6:1::1	sam0815-test.olsr		undefined	fdbe:924b:c48c:2::1
fdef:58c8:20a:1::1	AktiveZentren-Doerpfeldstr.olsr	undefined	fdbe:924b:c48c:2::1
fd08:2998:b87f:1::1	Ahof-frieden02.olsr		undefined	fdbe:924b:c48c:2::1
```
* `olsr.watchdog` file gets updated
* `127.0.0.1:9090`socket is not listening 

**good node: sam0815-test**
```
Neighbour IP 		Hostname 		Interface 	Local interface IP
10.230.140.32		Ahof-frieden03.olsr	wireless0	10.31.21.125
10.230.242.138		FF-TK-spare2.olsr	wireless0	10.31.21.125
fdbe:924b:c48c:2::1	Ahof-frieden03.olsr	undefined	fd2d:21b4:57d6:1::1
```

to sum it up:
* this wireless link seems ok, as IPv6 is passing 
* OLSR announcements gets sent from node `Ahof-Frieden03` but incoming announcements are ignored
* olsr.org - 0.9.0.3-git_1b6dc2e-hash_bf1cb86e7bc284f8fbd340a5681e692d from 'Freifunk Berlin kathleen 0.2.0-rc1+7644efc'
* `Nov 13 02:47 olsrd.pid` / `Nov 14 14:12 olsrd.watchdog`

sarumpaet, 2016-11-14:
Are you sure the IPv4 link is still working? I think I've seen IPv4 failing but IPv6 still being okay. Stracing olsrd might help, also looking at the logs, perhaps some interface disappeared temporarily, and olsrd failed to reopen it or something.

SvenRoederer, 2016-11-14:
tcpdump was (still) showing incoming packets from other olsr-nodes on wlan0-adhoc-2. 
logs did not show anything, indeed I had not extended the ring-buffer on this node yet.

I'll continue monitoring this.

SvenRoederer, 2016-11-14:
what about adding a label "olsr" to keep track of issues like this. So we can easy lookup them...

SvenRoederer, 2016-12-04:
seen again on "ahof-frieden03.olsr" (tpl-wr1043v2)
* olsrd is running for v4 and v6
  * olsrd is writing to each individual `watchdog-file`
  * port `9090` only for IPv6
* other nodes see this node
* IPv4 & IPv6 is working both ways for `ping` & `mtr`
* monitor.berlin.freifunk shows olsr-neigbors where lost over a week ago, so again no logs

bittorf, 2016-12-19:
olsr 0.9.5 is out - please think about including/testing this...

SvenRoederer, 2016-12-22:
@bittorf  seems that this causes some other deps (gps, updated LuCI) so maybe not an official build fro kathleen 0.2.0, but more after the switch to LEDE.