freifunk-berlin/firmware #407: Use LEDE instead of OpenWRT

niccokunzmann, 2016-12-01:
Using the [LEDE](https://www.lede-project.org/start) project should be easy. It is where the development goes to. Let us discuss it here.

lynxis, 2016-12-01:
look at the lede branch...

SvenRoederer, 2016-12-04:
as we are planning this for the next major-release I'll close this here

closed 2016-12-04 by SvenRoederer

SvenRoederer, 2017-01-02:
@niccokunzmann I put the commits we already have to support LEDE-build into a branch in my Repo (https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/build_with_lede) from this I added some "compatibility hacks" to build with current buildbot-setting (see https://github.com/freifunk-berlin/firmware/tree/SAm0815/build_with_lede-hack_for_buildbot).
This all is based on LEDE and stuff of end December 2016.

SvenRoederer, 2017-01-12:
See also #412