freifunk-berlin/firmware #408: CPE510 wifi low tx power no network scan on 0.2.0

bobster-galore, 2016-12-04:
1) There seems to be an issue with CPE510 on 0.2.0, the scan-button on network-wireless page produces a blank page.
2) Users report only 10mW tx-power as a client, whereas accesspoint and monitor produce 50mW.

bobster-galore, 2016-12-06:
1. in acccess point mode network scan works, in client mode it doesn't work.

bittorf, 2016-12-19:
maybe because of: https://git.lede-project.org/?p=source.git;a=commit;h=c2e0c41842895ba47819fa98b785c76a2524628b

booo, 2017-01-24:
> in access point mode network scan works, in client mode it doesn't work.

Scan works for me in client and ap mode in the gui and on the commandline.

> Users report only 10mW tx-power as a client, whereas accesspoint and monitor produce 50mW.

Which channel do you use? As client on the "right" channel I can configure e.g. 23 dBm at channel 149:

```
wlan0     ESSID: "OpenWrt"
          Access Point: 64:66:B3:C7:18:26
          Mode: Client  Channel: 149 (5.745 GHz)
          Tx-Power: 23 dBm  Link Quality: 70/70
          Signal: -25 dBm  Noise: -95 dBm
          Bit Rate: 13.0 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11an
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

You can check with `iw phy0 info` what is possible. Please check as well the regulations on the maximum output power and make sure you add the antenna gain to the output power.

Please (re)open a ticket with more information if you still have problems.

closed 2017-01-24 by booo