freifunk-berlin/firmware #409: owm partly crashes when olsr is disabled

bobster-galore, 2016-12-11:
kathleen 0.2.0

# owm.lua --dry-run
sh: write error: Broken pipe
{"freifunk":{"contact": ...

may be this gives a hint or is a second task
owm.lua --dryrun
sh: write error: Broken pipe
^C
/usr/bin/lua: /usr/lib/lua/luci/httpclient.lua:118: interrupted!
stack traceback:
    [C]: in function 'connect'
    /usr/lib/lua/luci/httpclient.lua:118: in function 'request_raw'
    /usr/lib/lua/luci/httpclient.lua:66: in function 'request_to_source'
    /usr/lib/lua/luci/httpclient.lua:54: in function 'request_to_buffer'
    /usr/sbin/owm.lua:29: in function 'db_put'
    /usr/sbin/owm.lua:88: in main chunk
    [C]: ?

SvenRoederer, 2016-12-11:
just had a quick test on LEDE, with owm-packages of kathleen-0.2.0, and all looks fine ... (no olsr running at all)
```
root@ALEP-eV-siemensstr:~# netstat -lnp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      791/uhttpd
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      996/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      710/dropbear
tcp        0      0 :::80                   :::*                    LISTEN      791/uhttpd
tcp        0      0 :::53                   :::*                    LISTEN      996/dnsmasq
tcp        0      0 :::22                   :::*                    LISTEN      710/dropbear
udp        0      0 0.0.0.0:53              0.0.0.0:*                           996/dnsmasq
udp        0      0 0.0.0.0:4500            0.0.0.0:*                           3532/charon
udp        0      0 0.0.0.0:35990           0.0.0.0:*                           5573/collectd
udp        0      0 0.0.0.0:500             0.0.0.0:*                           3532/charon
udp        0      0 :::546                  :::*                                777/odhcp6c
udp        0      0 :::53                   :::*                                996/dnsmasq
udp        0      0 :::4500                 :::*                                3532/charon
udp        0      0 :::500                  :::*                                3532/charon
raw        0      0 ::%1:58                 ::%4440556:*            58          777/odhcp6c
raw        0      0 ::%1:58                 ::%4440556:*            58          662/odhcpd
raw        0      0 ::%1:58                 ::%4440556:*            58          662/odhcpd
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node PID/Program name    Path
unix  2      [ ACC ]     STREAM     LISTENING       7742 3532/charon         /var/run/charon.ctl
unix  2      [ ACC ]     STREAM     LISTENING       1147 441/ubusd           /var/run/ubus.sock
root@ALEP-eV-siemensstr:~# owm.lua 
update Doc  Statuscode: 201 http://api.openwifimap.net//update_node/ALEP-eV-siemensstr.olsr (doc updated)
root@ALEP-eV-siemensstr:~# owm.lua --dryrun
update Doc  Statuscode: 201 http://api.openwifimap.net//update_node/ALEP-eV-siemensstr.olsr (doc updated)
root@ALEP-eV-siemensstr:~# 
```

sarumpaet, 2016-12-23:
Looks like an exception because there is no internet connection (or at least no connection to the server). Also happens on dry run because it's "--dry-run", not "--dryrun". Closing this for now.

closed 2016-12-23 by sarumpaet