freifunk-berlin/firmware #411: Variablenbelegung in  /etc/openwrt_release

AnnieYousar, 2017-01-04:
In /etc/openwrt_release werden einige Variablen belegt, die an verschiedenen Stellen verwendet werden:
DISTRIB_ID='Freifunk Berlin'
DISTRIB_RELEASE='0.2.0'
DISTRIB_REVISION='v0.2.0'
DISTRIB_CODENAME='kathleen'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='Freifunk Berlin kathleen 0.2.0'
DISTRIB_TAINTS='no-all busybox'

Die Variable DISTRIB_DESCRIPTION enthält dabei unnötigerweise die Release-Nummer. Die Release-Information wird durch DISTRIB_RELEASE und DISTRIB_REVISION bestimmt.
Mit anderen Worten 
DISTRIB_DESCRIPTION='Freifunk Berlin kathleen'
wäre völlig ausreichend.

AnnieYousar, 2017-01-04:
In /usr/lib/lua/luci/version.lua wird gestestet, ob DISTRIB_DESCRIPTION bereits die Zeichenkette DISTRIB_REVISION enthält. Wenn nicht, wird die interne Variable distversion aus DISTRIB_DESCRIPTION **und** DISTRIB_REVISION gebildet, ansonsten nur aus DISTRIB_DESCRIPTION. Wenn also DISTRIB_DESCRIPTION nicht geändert werden soll, dann wäre zur Vermeidung der unnötigen Dopplungen die Variable DISTRIB_REVISION='0.2.0', also ohne "v" zu setzen.

SvenRoederer, 2017-01-05:
Das ganze Benamsungszeuch ist in der Tat noch nicht ganz fertig. 
für's Kathleen-0.2.0-release sind extra die commits https://github.com/freifunk-berlin/firmware/commit/b7c6fd9964ce2a14911d76d20764ffecd07af42c & https://github.com/freifunk-berlin/firmware/commit/34c09dd9b8d8fe6703a3439fbc29c832381f3b5f gemacht worden, dass es albwerd paßt.

Hier ist noch arbeit nötig. wenn du praktische Vorschläge hast - sind wir für einen PR immer offen

AnnieYousar, 2017-01-05:
So wie ich das ganze jetzt verstanden habe, sollte in `openwrt_release` bei einer stable Version
```
DISTRIB_ID='Freifunk Berlin'
DISTRIB_RELEASE='0.2.0'
DISTRIB_CODENAME='kathleen'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='Freifunk Berlin kathleen 0.2.0'
DISTRIB_TAINTS='no-all busybox'
```
stehen. Die Variable `DISTRIB_REVISION=''` sollte dann leer bleiben.
Für eine unstable Version würde ich vorschlagen, entweder den Identifier `1167` aus dem Repository
(beispielsweise `/buildbot/unstable/ar71xx/1167/default/kathleen....`) oder einen passenden ISO-Zeitstempel, etwa `2017-01-05`, den das Kommando  `date -u -I'date'` erzeugt:
```
DISTRIB_ID='Freifunk Berlin'
DISTRIB_RELEASE='0.2.0'
DISTRIB_CODENAME='kathleen'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='Freifunk Berlin kathleen 0.2.0'
DISTRIB_REVISION='2017-01-05'
DISTRIB_TAINTS='no-all busybox'
```
Wenn das Datum zu ungenau sein sollte, könnte man mit `date -u -I'seconds'` auch einen genaueren Wert angeben.
Wann und durch wen wird eigentlich die Datei `openwrt_release` erzeugt?

SvenRoederer, 2017-01-05:
Am Donnerstag, 5. Januar 2017, 09:36:18 CET schrieb AnnieYousar:
> F�r eine unstable Version w�rde ich vorschlagen, entweder den Identifier
> `1167` aus dem Repository (beispielsweise
> `/buildbot/unstable/ar71xx/1167/default/kathleen....`) oder einen passenden
> ISO-Zeitstempel, etwa `2017-01-05`, den das Kommando  `date -u -I'date'`
> erzeugt:

f�r unstable build nutzen wir den GIT-hash. so ist der build auch immer 
eindeutig zu identifizieren.

AnnieYousar, 2017-01-05:
Wenn es darum geht, einen build zu identifizieren, dann ist ein kompletter Hash etwas "overdressed". Ich würde deshalb nur die ersten 8 Byte (die linken 16 Zeichen) nehmen. Ich glaube, dass wäre auch zulässig. Ergänzt man diese Folge vorn durch das Datum, dann lassen sich die Werte auch leicht sortieren:
`DISTRIB_REVISION='20170105-90474569d218cefb'`
Der verkürzte GIT-hash ist der vom LuCI build, der könnte übrigens auch kürzer sein ;-)

SvenRoederer, 2017-01-06:
Wenn du mal nachsiehst, erkennst du, dass der aktuell benutzte git-hash der üblichen 7byte-Konvention entspricht.
Bisher haben wir keine Notwendigkeit gesehen, ein Datum im buildstring zu haben.

Wenn dir der LuCI-build-string nicht gefällt, möchte ich dich gern an die LuCI-entwickler verweisen (https://github.com/openwrt/luci).

SvenRoederer, 2017-01-08:
ja, 7 byte.
Weil's halt einkompiliert wird, nehmen wir den GIT-hash zur compile-time. Somit ist auch eindeutig klar, aus welchem Code-stand die firmware gebaut ist. Im gegensatz zum Datum kann man so auch noch mach 1 Jahr eine Firmware mit dem selben Stand und entsprechendem Namen bauen.

SvenRoederer, 2017-01-08:
to come back to the original issue: maybe we should check on the next OpenWrt-based release if we should avoid touching REVISION in $(OPENWRT_DIR)/include/version.mk (
https://github.com/freifunk-berlin/firmware/blob/0.2.0/Makefile#L105)

For me the configured defaults at buildtime (https://github.com/freifunk-berlin/firmware/blob/340a94cf52bdaf20f9606c1272cffc5236f91d67/configs/common.config#L175) looking fine

AnnieYousar, 2017-01-15:
Sven,
I see that the problem is solved in the LEDE migration branch. For convienence I attached the /etc/banner and /etc/openwrt_release contents related to the issue:
In kathleen 0.1.2 we had: `Firmware Berlin (kathleen 0.1.2, based OpenWRT r44952)`
in kathleen 0.2.0 it changes to: `Firmware Berlin (kathleen 0.2.0 rev v0.2.0)` with the doubled rev
and in LEDE migration now: `Firmware Berlin (LEDE-migration 38ce76d rev r2961-5b089e4)`
Although I don't understand yet, where the 7char strings are got from, the DISTRIB_REVISION in /etc/openwrt is now set accordingly. Thanks, the issue is closed.
Kind regards,
/Ann.
BTW the first dashed line in the banner file disappered in LEDE. Was that intentionally made?
[diff banner 0.1.2 0.2.0 and LEDE.txt](https://github.com/freifunk-berlin/firmware/files/706507/diff.banner.0.1.2.0.2.0.and.LEDE.txt)

closed 2017-01-15 by AnnieYousar

SvenRoederer, 2017-01-15:
The LEDE-branch is still very hackisch. This minor thing I've seen also.
On such "switch-over problems" we will work, when there is a agreed roadmap to migrate over.