freifunk-berlin/firmware #414: save some disc-space when building on buildbot

SvenRoederer, 2017-01-08:
Think we can save some disc-space on buildslaves and makes buildprocess more nice by:
- building OpoenWrt / LEDE
- copy all files we need to build our firmwarefiles (at least: imgbuilder, packages; in addition SDK)
- call ```make clean``` in upstream-firmware (this will remove ``´bin/``` and ```build_dir/target*```)

for better reading, we can also move the "Version.txt" section into a separate make-target

SvenRoederer, 2017-01-08:
the ```make clean``` we should per default only do on buildbot-builds, to keep the binaries when compiling locally.
PoC for detection is https://github.com/freifunk-berlin/buildbot/pull/42

SvenRoederer, 2017-03-23:
recent OpenWrt builds failing on slave wg1337, see https://buildbot.berlin.freifunk.net/builders/ar71xx-generic/builds/175

SvenRoederer, 2017-04-17:
still valid for LEDE 17.01 (https://buildbot.berlin.freifunk.net/builders/ar71xx-generic/builds/214)

bobster-galore, 2017-09-12:
Will there be a new behaviour? Gives moving files enough room for the builds?

SvenRoederer, 2017-09-14:
In 64dfaea the code of PR #452 was merged, which moves some files instead of copying them.
This saves probably 300 MB, but buildslave "wg1337" still fails when building target ar71xx-generic.

SvenRoederer, 2017-10-08:
What about removing ar71xx-generic from the list of valid targets for buildslave wg1337?

SvenRoederer, 2017-12-11:
https://github.com/freifunk-berlin/firmware/commits/issue414_v2 is an alternative approach to handle this.

SvenRoederer, 2018-02-06:
This was autoclosed, but still some working detection for the buildbot is missing. --> https://github.com/freifunk-berlin/buildbot/pull/42

SvenRoederer, 2018-03-22:
I pushed "Makefile: detect if we running on a special host" to branch next_release in order to merge into master.
This problem is known since >1year and there was only one comment, on this solution. It was that it's bad to have different rules / execution of the Makefile when running on Buildbot or having a local build. But this can be avoided by setting "SET_BUILDBOT" in config.mk (https://github.com/freifunk-berlin/firmware/commit/ec06e30ea452150e75b3e45ff23efa2a6d623200)

SvenRoederer, 2018-04-27:
so this code made it into master and Hedy-1.0.0

closed 2018-04-27 by SvenRoederer