freifunk-berlin/firmware #415: neigh.sh missing in the latest stable release

booo, 2017-01-12:
In the latest stable release (backbone) the neigh.sh script is missing. The script is quite handy for debugging purposes and we should check why it's missing.

SvenRoederer, 2017-01-12:
> root@Ahof-frieden03:~# opkg search /usr/bin/neigh.sh 
freifunk-common - 1

I assume a missing dependency in freifunk-mod-ui

SvenRoederer, 2017-01-12:
using the current feeds of LuCI (https://github.com/freifunk-berlin/firmware/tree/SAm0815/build_with_lede) it reappears in the backbone-image by default

sarumpaet, 2017-01-13:
freifunk-common is another candidate for a fork I guess. For example, we probably don't want stuff like https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-common/files/etc/init.d/freifunk (which we also partially duplicate in https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-dhcp-defaults/files/etc/init.d/ff_dnsmasq_reload ...).

SvenRoederer, 2017-01-13:
@sarumpaet indeed this script clashes with our berlin*default packages.

but instead of forking, we can maybe introduce a UCI-setting (freifunk.run_init) where this script can be turned off?

booo, 2017-01-14:
Maybe we should split up the freifunk-common package or create the possibility to select which files from the package we actually want (e.g. neigh.sh and maybe the ff_olsr_watchdog)?

sarumpaet, 2017-01-14:
@SvenRoederer Hm, of course we could introduce a setting, but what would that gain us? We'd have another obscure dependency/package interaction, and the fact that "freifunk-common" is misnamed, mixing things that don't belong together, and de facto not maintained doesn't go away.

We could move the watchdog to freifunk-berlin-olsrd-defaults, and create a freifunk-(shell/cli-)utils package or similar for the shell scripts; then we can also easily add any new own scripts.

SvenRoederer, 2017-01-14:
regarding blaming freifunk-common: the interesting point is, that the makefile did not change since 2011 and the last change in the neigh.sh-script was done in Nov.2015.

So there was no change in the deps.

Any ideas why it wasn't put in the image? Feels to me, that's a fault on our side. Also have in mind, that's included in the default-image ...

FreifunkUFO, 2017-01-18:
@SvenRoederer last change in the neigh.sh script? its only some weeks ago (because of olsr changes in version 0.9.5)! or do you have a new built berlin-script for that with same name?

https://github.com/openwrt/luci/commits/master/contrib/package/freifunk-common/files/usr/bin/neigh.sh

SvenRoederer, 2017-01-18:
@FreifunkUFO good, to make my comment more precise: the last change to neigh.sh before the release of kathleen-0.2.0.
Changes to the script after the firmware-release, are cearly not included :-)

SvenRoederer, 2017-03-19:
have you checked 0.3.0-rc1 already?

booo, 2017-03-19:
It's part of this build: http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips-mt7621/125/backbone/

I think the problem is solved. If someone wants to reorganize the package please open another ticket.

closed 2017-03-19 by booo