freifunk-berlin/firmware #416: enable NAT on ffvpn interface for IPv4

booo, 2017-01-12:
We should enable NAT on the ffvpn interface.

In our current setup we do not translate addresses that are routed via the ffvpn interface of the router to the vpn servers. We do NAT on the vpn servers.

For this setup we have to maintain a patched version of openvpn for the servers. Maintaining this patched version of openvpn is a real pain. Without the patch openvpn does not know how to route data to nodes that are not directly connected to the vpn server as vpn clients. The patch enables openvpn to learn about the nodes that are reachable via a specific vpn client by inspecting the src addresses of packets.

By enabling NAT on the routers vpn interface we can drop the patch because openvpn only has to keep track of the vpn clients ip and it does this out of the box.

The change will enable users to use other vpn services without making big changes to the vpn and network configuration on the router.

Enabling NAT on the router may help make the transition to other vpn implementations easier too.

booo, 2017-01-12:
Maybe we just have to add this to the firewall configuration:
```
config zone ffvpn       
        option name 'ffvpn'
        option network 'ffvpn'
        option forward 'ACCEPT'                                                                                                       
        option output 'ACCEPT' 
        option input 'ACCEPT'
        option masq '1'
```

sarumpaet, 2017-01-13:
There used to be this thing in the config files:
https://github.com/freifunk-berlin/firmware-packages/blob/c9c97de959c3d6fb4394fe4e028b19ca87a84520/utils/freifunk-berlin-firewall-defaults/uci-defaults/freifunk-berlin-firewall-defaults#L131
AFAIR it was enough to use MASQUERADE_TNL instead of FORWARD_TNL to make it work.
I guess NAT will happen on all nodes then (before the SGW tunneling), not only on VPN enabled nodes, which might be a good thing...?

Also, perhaps @sven-ola and/or @basros can explain why NAT currently happens on the VPN server?

booo, 2017-01-14:
If we enable NAT on the router we still have to use NAT on the servers. We don't (I guess) have enough public ip space to assign a public ip to very router. Enough public ip space would allow us to push a public ip from the server to every router/gateway and route the data without address translation from the server to the gateway.

If we enable NAT on the router we translate the address two times. Rumor has it that this brakes some applications e.g. VoIP.

The configuration for the tunnel interfaces seems fine to me. We do not want to masquerade the address when we send data through the ipip-tunnel of the sgw. In our private network there is no reason to translate addresses. We can route everything.

I think there is no masquerading without forwarding.

SvenRoederer, 2017-01-30:
regarding "only NAT on gateways": this might be related to the "zap-script".

booo, 2017-01-31:
@SvenRoederer Why?

SvenRoederer, 2017-02-02:
https://wiki.freifunk.net/Vpn03_Admin_Howto#Die_Sache_mit_dem_ZAPP with NAT on the Nodes the gateway can only see the connections of all clients of a node and limit them all

booo, 2017-02-02:
The zapp-script would see more connections from one IP and could only block connections from the router IP. Right. But the zapp-script is disabled for quite some time now.

SvenRoederer, 2017-10-09:
implemented with https://github.com/freifunk-berlin/firmware-packages/commit/2095340367da5be49ac227b2bce87ba5c203b9ac

closed 2017-10-09 by SvenRoederer