freifunk-berlin/firmware #418: issues seen with LEDE-base build

SvenRoederer, 2017-01-19:
this is just fir collecting roughly the isses seen when building with LEDE-17.01-branch. Should be transformed into separate issues as soon as we switch to LEDE, or discard this issue when LEDE is not an option.

SvenRoederer, 2017-01-19:
* ff-assistent: requires setup of "private AP" to continue
* ff-assistent: shows textbox with OpenVPN-key paths inside
* default 4MB-images only build for TPL-WR701, other rootfs are to big
* buildbot "wg1337" fails, as of low disk-space

bobster-galore, 2017-01-20:
- on  cpe210 owm.lua doesn't work
- on  cpe210 olsrd isn't stable, forgets neighbours names or routes to them
- on  cpe210 olsrd doesn't know it's cpe210.olsr only ipv4-address works

bobster-galore, 2017-01-24:
- on cpe210 no data for hopglass & openwifimap while monitor is up to date

booo, 2017-01-24:
> on cpe210 no data for hopglass & openwifimap while monitor is up to date

If owm.lua does not work/report data hopglass & openwifimap does not have the data for visualization.

pmelange, 2017-02-05:
Based on branch SAm0815/build_with_lede:

The libraries referenced in /etc/config/olsr and /etc/config/olsr6 need to be updated.
olsrd_txtinfo.so.0.1 -> olsrd_txtinfo.so.1.1
olsrd_nameservice.so.0.3 -> olsrd_nameservice.so.0.4
olsrd_jsoninfo.so.0.0 -> olsrd_jsoninfo.so.1.1

The luci status page for configured interfaces does not show the actual data.
The luci status page for the smart gateway shows nothing although the plugin definately sets up a tunnel correctly.

Running a command such as 'echo "/gateways" | nc 127.0.0.1 9090' is not nicely indented and organized like it is in 0.2.0.  There is also some new information provided by this command.  Perhaps that is having in influence on the luci status pages.

pmelange, 2017-02-05:
Based on branch SAm0815/build_with_lede:

I disabled the 'berlin.freifunk.net' SSIDs on both 2.4Ghz and 5Ghz to prevent clients from attaching to my test node.  On the luci wireless status page, the SSID's shown for the disabled interfaces are not showing up as 'berlin.freifunk.net' but as 'intern-chXX.freifunk.net'.

SvenRoederer, 2017-02-05:
@pmelange the build_with_lede branch is using recent openwrt-packages, which has updated olsrd v0.9.6. this causing new plugin-versions. Upstream LuCI integration needs to be tested.

Most bad to me seems, that OLSRs 0.9.6 had very unstable Routing on my last short test. So I created branch "build_with_lede_and olsr0.9.0.3" this uses olsrd as used in kathleen-0.2.0.
Further test needs to be done on olsr to eventually open ticket on upstream

pmelange, 2017-02-05:
@SvenRoederer Can you be more descriptive about the routing problems?  

I'm meshing onto the backbone directly, my uptime is just over 9hrs, my tunnel has been built (and changed 7 times) and packets are reaching their destinations.

root@perry-test:~# ip r s t olsr | wc -l
949

SvenRoederer, 2017-02-12:
reported upstream via https://github.com/OLSR/olsrd/issues/20

bobster-galore, 2017-03-18:
hint: https://github.com/openwrt/luci/blob/fecf6e1d11d466efa7b1bbf05633931a3081517e/modules/luci-base/luasrc/tools/status.lua#L160

SvenRoederer, 2017-03-19:
Hi all, please check if any of the listed Issues is still relevant and open an individual Issue for it. 

I'd like to close this soon.

SvenRoederer, 2017-04-09:
it was discussed, that al reported issues have been moved to separate issues or are solved or not relevant anymore

closed 2017-04-09 by SvenRoederer