freifunk-berlin/firmware #419: CPE510 - missing outdoor channels

xa-y-ax, 2017-01-25:
5770 MHz (DFS)	5775 MHz (DFS)	5780 MHz (DFS)	5785 MHz (DFS)
5790 MHz (DFS)	5795 MHz (DFS)	5800 MHz (DFS)	5805 MHz (DFS)	5810 MHz (DFS)
5815 MHz (DFS)	5820 MHz (DFS)	5825 MHz (DFS)	5830 MHz (DFS)	5835 MHz (DFS)
5840 MHz (DFS)	5845 MHz (DFS)	5850 MHz (DFS)	5855 MHz (DFS)	5860 MHz (DFS)
are not available in kathleen 0.2.0 on a TP-LINK CPE 510.

bobster-galore, 2017-01-26:
Did you drop the code snippet by purpose?

xa-y-ax, 2017-01-26:
David send this snippet to fix the problem but Sven said it probably only influences the UI - so I dropped it...

```

--- net/wireless/util.c.orig    2015-01-08 10:31:04.175732216 +0000
+++ net/wireless/util.c    2015-01-08 10:35:16.905572659 +0000
@@ -49,6 +49,8 @@
    case IEEE80211_BAND_5GHZ:
        if (chan >= 182 && chan <= 196)
            return 4000 + chan * 5;
+        else if (chan > 196)
+            return 5000 + (chan - 15) * 5;
        else
            return 5000 + chan * 5;
        break;
@@ -72,6 +74,8 @@
        return (freq - 2407) / 5;
    else if (freq >= 4910 && freq <= 4980)
        return (freq - 4000) / 5;
+        else if (freq >= 5910)
+        return (freq - 5000) / 5 + 15;
    else if (freq <= 45000) /* DMG band lower limit */
        return (freq - 5000) / 5;
    else if (freq >= 58320 && freq <= 64800)
```

bobster-galore, 2017-01-26:
May be we find useful information at openwrt ticket https://dev.openwrt.org/ticket/19402 it's about frequencies and tx-power. Unfortunately ends with the revoked tx-power patch.

SvenRoederer, 2020-05-23:
This is the LuCI page for WiFi setup taken on a CPE510v1 running at leaat our firmware based on OpenWrt-19.07.2 (5bd10142dd2e3534e7928da6ad6e8b9321f438ab I assume)

[image]

Channels up to 165 are avail. Not checked the iwinfo output.