freifunk-berlin/firmware #423: olsrd2 error on activating init scripts

lynxis, 2017-02-18:
```
/home/freifunk/firmware/imgf7aweQ/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-generic.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/init.d/dlep_radio: line 6: /lib/functions/oonf_init.sh: No such file or directory
/home/freifunk/firmware/imgf7aweQ/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-generic.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/init.d/olsrd2: line 6: /lib/functions/oonf_init.sh: No such file or directory
rm -f /home/freifunk/firmware/imgf7aweQ/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-generic.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/usr/lib/opkg/info/*.postinst
make[3]: Leaving directory '/home/freifunk/firmware/imgf7aweQ/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-generic.Linux-x86_64'
```

SvenRoederer, 2017-02-19:
i've seen this also elsewhere:
```
Activating init scripts
/var/lib/buildbot/slaves/slave/ar71xx-mikrotik/build/firmware/imgrctMnb/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-mikrotik.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/init.d/freifunk-policyrouting: line 4: /lib/functions/network.sh: No such file or directory
/var/lib/buildbot/slaves/slave/ar71xx-mikrotik/build/firmware/imgrctMnb/freifunk-berlin-imagebuilder-0.3.0-alpha-ar71xx-mikrotik.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx/etc/init.d/freifunk-policyrouting: line 5: /lib/functions.sh: No such file or directory```
```

bobster-galore, 2018-03-30:
Can we use olsrv2 on Hedy? Has there been any progress?

SvenRoederer, 2018-04-01:
olsrdv2 ist in the packages-repo since at least v0.2.0.
binary packages for babel, bmx 6 & 7 are included in the SAm0815_experimental branch for now

SvenRoederer, 2018-04-01:
regarding the similar issue I mentioned before (https://github.com/freifunk-berlin/firmware/issues/423#event-968341976) I opened a PR with a fix https://github.com/openwrt/luci/pull/1635

SvenRoederer, 2020-02-16:
freifunk-policyrouting has been tracked upstream upstream by: https://github.com/freifunk/openwrt-packages/issues/10 and was fixed by https://github.com/freifunk/openwrt-packages/commit/5425bc9daa0cf18a9aff6e77c3b35c677e04f538.