freifunk-berlin/firmware #424: olsrd: v0.9.5 crashes when routes come in

SvenRoederer, 2017-02-19:
current olsrd v0.9.5 crashes as soon routes come in from other neighbours.Console shows some kind of '''ASSERT''' fails. Don't know the exact message or code anymore.

When using current olsrd 0.9.6+ all is fine again

SvenRoederer, 2017-02-22:
@fhuberts I pull the discussion on this issue out of olsrd/#20  to not mix all together. Shall I open an separate issue in olsrd-repo for this too?

git bisect on olsrd for this issue ends in
```
sven@build:/usr/src/network/olsrd$ git bisect bad 2e7f2942bd47fc7f0b4ca0bf458
2e7f2942bd47fc7f0b4ca0bf4581c3dee3c1f85a is the first bad commit
commit 2e7f2942bd47fc7f0b4ca0bf4581c3dee3c1f85a
Author: Ferry Huberts <ferry.huberts@pelagic.nl>
Date:   Mon Nov 23 09:00:08 2015 +0100

    sgw: only select an olsr or egress interface when it is up
    
    Signed-off-by: Ferry Huberts <ferry.huberts@pelagic.nl>

:040000 040000 886fd9e696469edb0a547757957c9bd38a61bb1e 45f928c58d66d7f0042427b2d9e22ea11a26bba5 M      src
```
here the log:
```
git bisect start
# bad: [4333a8b9c7f1f156ea457168837163a8d17685fb] Release v0.9.5
git bisect bad 4333a8b9c7f1f156ea457168837163a8d17685fb
# good: [c6fbdafd11ef1d31cbbeab138317c3fdd6673d1a] Release v0.9.0.3
git bisect good c6fbdafd11ef1d31cbbeab138317c3fdd6673d1a
# bad: [0d460d5fe4dcf0fc482f5f0294086ed8f4a304f0] jsoninfo: minor improvements
git bisect bad 0d460d5fe4dcf0fc482f5f0294086ed8f4a304f0
# bad: [ffaa7324a2355f12551a351366d0236e6a4d1532] txtinfo: fixup the Content-Type http header
git bisect bad ffaa7324a2355f12551a351366d0236e6a4d1532
# good: [49a9ea67f110814e3a7af9642cd25e17df2ffbf0] main: move some initialisation to the beginning of main
git bisect good 49a9ea67f110814e3a7af9642cd25e17df2ffbf0
# bad: [4e58755a1d3c75a656851873c7eab23c304049c6] jsoninfo: sgw field in /sgw response is not an array
git bisect bad 4e58755a1d3c75a656851873c7eab23c304049c6
# good: [42a0da35ab9261cbe78a8f2b3b495dc9636f3663] main: minor update
git bisect good 42a0da35ab9261cbe78a8f2b3b495dc9636f3663
# good: [34285e6b10430a7ce8f836937aa4f3fe131261bc] conf: avoid memory leaks during parsing
git bisect good 34285e6b10430a7ce8f836937aa4f3fe131261bc
# good: [84b43d879595ffda7e1ae22cb5eb8dc0e0daa903] conf: olsrd_free_cnf: reorder a bit
git bisect good 84b43d879595ffda7e1ae22cb5eb8dc0e0daa903
# good: [55a42f60552e8b7b35cfed439a65fe48e7d3ffd0] conf: initialise sgw uplink & downlink before setting them
git bisect good 55a42f60552e8b7b35cfed439a65fe48e7d3ffd0
# good: [2e568eb7264dd9df3fcf68db835b066adee6546f] main: minor update
git bisect good 2e568eb7264dd9df3fcf68db835b066adee6546f
# bad: [2e7f2942bd47fc7f0b4ca0bf4581c3dee3c1f85a] sgw: only select an olsr or egress interface when it is up
git bisect bad 2e7f2942bd47fc7f0b4ca0bf4581c3dee3c1f85a
# first bad commit: [2e7f2942bd47fc7f0b4ca0bf4581c3dee3c1f85a] sgw: only select an olsr or egress interface when it is up
```

fhuberts, 2017-02-22:
That one was fixed by 97d491648b26fc5550c27fb3a82eaacfe6472325:

```
Author: Ferry Huberts <ferry.huberts@pelagic.nl>  2016-12-01 18:08:14
Committer: Ferry Huberts <ferry.huberts@pelagic.nl>  2016-12-05 16:41:10
Parent: 85d249d60261ea29e21de33501a6df61dfb1a259 (release: adjust script to create multiple tar files)
Child:  3f7c2136c2f86ce3edbebd26f9921f0a6fb1e039 (Merge branch 'release-0.9.5')
Child:  93b70d54f09c730560feb48bd258d3b268f8c467 (txtinfo: do not use http headers by default)
Branches: master, remotes/olsr/master, remotes/origin/master, remotes/public/master, remotes/stinkpad/master and many more (22)
Follows: v0.9.5
Precedes: v0.9.6

    sgw: fix interface monitor for single-sgw mode
    
    Signed-off-by: Ferry Huberts <ferry.huberts@pelagic.nl>
```

So an upgrade to the latest will fix that.

SvenRoederer, 2017-03-19:
we will use OLSRd 0.9.6 which is (soon) the current release and is not affected y this anymore.

closed 2017-03-19 by SvenRoederer

fhuberts, 2017-03-19:
please use v0.9.6.1