freifunk-berlin/firmware #427: Private Access Point required

niccokunzmann, 2017-03-04:
```
 Firmware Berlin (Hedy 0.3.0-alpha rev eab0343)
 Generic - ramips/rt305x
```
This is the self-compiled software, without vpn.
When I access the GUI, Freifunk-assistent, I am required to type in a private access point password even when I do not make a tick at "private access point".
I would expect to not see and need to do anything when the private access point option is not chosen.

(The only password that worked for me was "00000000")

SvenRoederer, 2017-03-04:
Thanks for opening this issue. I've seen this already an noted in #418 .

It must relate to updated codebase of LuCI, that comes with the current LuCI-feed.

SvenRoederer, 2017-03-04:
I vote for removing this private AP feature at all, as once discussed in #143 .

SvenRoederer, 2017-03-06:
@niccokunzmann I reverted some commits to exclude the AP-feature again.
CAn you please try https://buildbot.berlin.freifunk.net/buildbot/unstable/mpc85xx-generic/143/packages/packages/powerpc_8540/packages_berlin/luci-app-ffwizard-berlin_0.0.6-1_all.ipk

niccokunzmann, 2017-03-06:
@SvenRoederer What can I do with the ipk file?
I would expect to pull and checkout a git commit to test changes.

SvenRoederer, 2017-03-06:
@niccokunzmann hope you have a spare-router, where you can upgrade this ffwizard-package and then start the configuration.

you can for sure compile by yourself, just follow the branch the mentioned commit belongs to.

SvenRoederer, 2017-06-02:
see PR https://github.com/freifunk-berlin/firmware-packages/pull/107 to remove this code

booo, 2017-08-11:
We removed the private ap code as discussed early.

closed 2017-08-11 by booo