freifunk-berlin/firmware #428: image too big - how to remove and add vpn

niccokunzmann, 2017-03-05:
I always get the warning and make error that the image is too big when I compile
```
WARNING: Image file /home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin is too big
/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/staging_dir/host/bin/mkporayfw -B A5-V11 -F 4M -f /home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin -o /home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin.new
[mkporayfw] *** error: stat failed on /home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin:No such file or directory
Makefile:254: recipe for target '/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin' failed
make[4]: *** [/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_rt305x/tmp/freifunk-berlin-0.3.0-alpha-eab0343-ramips-rt305x-a5-v11-squashfs-factory.bin] Error 255
make[4]: Leaving directory '/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64/target/linux/ramips/image'
Makefile:172: recipe for target 'build_image' failed
make[3]: *** [build_image] Error 2
make[3]: Leaving directory '/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64'
Makefile:110: recipe for target '_call_image' failed
make[2]: *** [_call_image] Error 2
make[2]: Leaving directory '/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64'
Makefile:199: recipe for target 'image' failed
make[1]: *** [image] Error 2
make[1]: Leaving directory '/home/user/firmware/imgEkVU9q/freifunk-berlin-imagebuilder-0.3.0-alpha-ramips-rt305x.Linux-x86_64'
Makefile:126: recipe for target '.stamp-firmwares' failed
make: *** [.stamp-firmwares] Error 2

user@ubuntu:~/firmware$ cat config.mk 
# default parameters for Makefile
SHELL:=$(shell which bash)
TARGET=ar71xx-generic
PACKAGES_LIST_DEFAULT=default_4MB
LEDE_SRC=https://git.lede-project.org/source.git
LEDE_COMMIT=ac733df99c78f54b4cf9a8710f70a506f73e66fc
MAKE_ARGS=

user@ubuntu:~/firmware$ cat profiles/ramips-rt305x.profiles 
a5-v11

user@ubuntu:~/firmware$ cat configs/ramips-rt305x.config 
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_rt305x=y
CONFIG_TARGET_ramips_rt305x_Default=y
CONFIG_PACKAGE_kmod-ath9k-htc=n
CONFIG_PACKAGE_kmod-mt76=n
CONFIG_PACKAGE_kmod-rt2800-lib=n
CONFIG_PACKAGE_kmod-rt2800-usb=n
CONFIG_PACKAGE_kmod-rt2x00-lib=n
CONFIG_PACKAGE_kmod-rt2x00-usb=n
CONFIG_PACKAGE_kmod-usb-dwc2=n
CONFIG_PACKAGE_kmod-usb2=n
#CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=128
```

The goal: I want to compile the freifunk firmware for the [3g router](https://wiki.openwrt.org/toh/unbranded/a5-v11). I also want to include the VPN.
However, including the VPN makes the image too large.

Questions:

- Is there a way to find out how large it is and how large it is supposed to be?
- How can I reduce the image size even further without risking to not be able to flash again through web/ssh?
- Should/can we reduce the size of the openvpn (we do not need tcp, some other options)

SvenRoederer, 2017-03-05:
that's a know fact. please check:
* https://github.com/freifunk-berlin/firmware/commit/785c9bde58a063b2eb2be9e2b9b8d68342a7ba45
* https://github.com/freifunk-berlin/firmware/commit/827d03c43e1429b20019da8475f2ac4c43afa957

so getting all this into the 4MB-image will never work

niccokunzmann, 2017-03-06:
Thanks! I am closing the issue now, as it can not be resolved.

niccokunzmann, 2017-03-06:
Would it be possible to merge openvpn and openssl and throw  out everything we do not need to create a very small openvpn+openssl binary?

AnnieYousar, 2017-03-06:
> that's a known fact. please check:
>   * 785c9bd
>   * 827d03c

But that's only a workaround and not the solution.
The better way is to issue RSA-2048 certs ;-)
/Ann.

booo, 2017-03-06:
@AnnieYousar What about the old certificates that are in use already?

If openvpn + mbedtls is the main cause for the disk space problem we could try to patch mbedtls to accept our certificates again or use an older version of mbedtls.

AnnieYousar, 2017-03-06:
@booo: First, start to issue certs with RSA2048 **and** SHA-256 certs (if not already ECC based). 
If it works, mail to 
`for c in *.crt; do  openssl x509 -in $c -subject -noout | sed 's#^.*emailAddress = ##;s#,.*##'; done`
the new issued certs. 
I'll post a shell script for re-issuing the certs later.

AnnieYousar, 2017-03-06:
Note that the current freifunk CA cert has a  NotAfter of Nov 26 21:00:14 2022 GMT. So you have to re-issue certs anyway until this date.

I made from scratch some scripts that support re-issuing:
- start the initialisation of a directory, use here the given `rsa_init_ca.sh` and `freifunk.cnf`, a config file for openssl, have a look in the file, it creates CA key, CA file and some other openssl stuff
- make a list from all old certs, extracting the commonName and the emailAddress, use e.g., a loop as given above
- execute the "list" with the help of the user cert generation script `nutzer_rsakey.sh`, it generated a key, creates a cert and makes a tgz, an example file `_liste.sh` is given in the attachment.

A medium PC generates one cert in 2 seconds, so it finishes in 3 days for 2000 certs. If you polish the scripts it may be even faster.
[freifunk.zip](https://github.com/freifunk-berlin/firmware/files/821482/freifunk.zip)

SvenRoederer, 2017-03-06:
@niccokunzmann creating a separate image-type might work, but w/o
* LuCI --> only ssh
* maybe OLSR needs to be excluded as well

try yourself by creating a setup in "packages/VPN03-commandline-only.txt". Also https://github.com/freifunk-berlin/firmware/commit/827d03c43e1429b20019da8475f2ac4c43afa957 needs to be reverted.

SvenRoederer, 2017-03-06:
I personally will not invest in this
* VPN03 is planned to shutdown (by the end of 2017)
* this only affects the 4MB-Routers which have a warning for a long time
This applies to both re-certifying and reenabling RSA1k in mbedTLS

SvenRoederer, 2017-03-06:
@AnnieYousar thanks for the scripts, but the CA-code is mostly ported to python (https://github.com/freifunk-berlin/ca.berlin.freifunk.net)
so we will unlikely rebuild a CA based on shell-skripts. The direction is https://github.com/freifunk-berlin/ca.berlin.freifunk.net/pull/53

AnnieYousar, 2017-03-06:
Are solar powered routers with lower power consumption out of scope now?

SvenRoederer, 2017-03-06:
e.g. "GL.iNet AR-150" and "Zsun wifi card reader" are supported / known to work :-)

sarumpaet, 2017-03-06:
We still have about 100 4MB devices on the network, which makes that more than 10% of the Freifunk Berlin devices. I don't agree with dropping support. Apart from the certificate re-issue idea, there are probably plenty of other possible solutions (e.g., building a static small openvpn version, and pushing that into the 4MB images using the default_4MB.sh hack script). Also, isn't the 2048 bits requirement in mbedtls just a one-liner? See here: *https://github.com/ARMmbed/mbedtls/blob/mbedtls-2.0/library/x509_crt.c#L98* - we could simply patch that (and should, as OpenSSL is quite a mess and needlessly large. If I'd spotted that commit in the heap of other LEDE-related changes, I'd vetoed it).

I don't see any problem with 1024 bits keys. VPN03 never was about encryption, and post-auth encryption is off anyways, so mentioning VPN03 in context with encryption is actually quite misleading. As I said multiple times already, "VPN03" was a misnomer really, it should have been SOMETHING-tunnel or similar ("fftunnel4lawyers"?), not any word commonly associated with encryption.

SvenRoederer, 2017-03-06:
Nobody really was planning to drop 4MB at all.
When I focussed on the LEDE-switch (based on @lynxis) I focussed on getting it running at all and tackle / find the major issues. As 4MB always failed and is a minor thing I just disabled it by a simple commit, which can be reverted anytime.
When looking ahead and seeing the upcoming VPN-changes I suggest to finish the LEDE-switch for v1.0.0 for >=8MB. For v1.0.1 there sholud be a vision what kind of tunnel or even if any kind of tunnel will be used. And this vision should take size of network-releted components into account. Than lets reenable 4MB.

SvenRoederer, 2017-03-19:
closing for now, as nobody seems to dislike this planning

closed 2017-03-19 by SvenRoederer

SvenRoederer, 2018-01-02:
@torte71 I heard that you are interested in broaden the use of the 4MB devices beyond the flavors "default" (aka no-tunnel) and "backbone".

here some ideas:
* https://github.com/freifunk-berlin/firmware/issues/428#issuecomment-284536889: downloading packages to RAM at runtime
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/my_routers/packages/mesh.txt: an 4MB compatible image, w/o webinterface, but olsrv1 and B.A.T.M.A.N.

torte71, 2018-01-02:
@SvenRoederer : The description for "mesh.txt" should rather read: -with- webinterface (but uplink only via mesh)

My idea was removing lua+uhttpd+luci. This gives enough room for openvpn-mbedtls (and more), so it could use the tunnel-berlin as uplink (I guess, many of the 4mb boxes don't have a mesh-partner in reach). BBB needs to be resolved somehow, as openvpn-openssl is still too big. And a bash-script that replaces at least the wizard would be a nice-to-have to not scare away noobs completely (wasn't there even a hackathon attempt to replace LuCi with a browser-based configurator that is hosted outside the router?)

The idea of downloading packages into RAM at runtime is promising, but we need to take care of security: If an attacker can replace the packages on the server, they would be automagically installed to many devices, giving root access to the internal networks. So at least a checksum test with a decent algo (rather sha256 than md5) is a must.
What about RAM-size? In discussions, I've been told, that the wr841 ist considered unstable, because it rebooted due to missing RAM when two people were administering it at the same time (though that contradicts my experience, my 841 boxes run rock-solid, even the one with vpn03, bbb-vpn and a private AP running at the same time).

Anyway, I think this issue-tracker isn't exactly the best place for brainstorming, a possibility for more structure, e.g. a Wiki or Pad page would be more suitable. imho.

SvenRoederer, 2018-01-02:
* "mesh.txt" is intended to be a full meshing-node, w/o direct uplink. And it's already w/o web-components.
* if you can package "opkg", it will take care of the checksum / integrity
* RAM-usage is the other side of the coin ...

Needs to be decided after some test.
For documenting we have in Github:
* https://github.com/freifunk-berlin/firmware/projects
* https://github.com/freifunk-berlin/firmware/wiki

SvenRoederer, 2018-01-03:
> wasn't there even a hackathon attempt to replace LuCi with a browser-based configurator that is hosted outside the router?

there is the branch "https://github.com/freifunk-berlin/firmware/tree/lede-wizard" (using https://github.com/freifunk-berlin/firmware-packages/tree/master/utils/freifunk-berlin-wizard-backend and https://github.com/freifunk-berlin/firmware-packages/tree/master/utils/freifunk-berlin-wizard-frontend) but as I remember even this was to big for 4MB devices