freifunk-berlin/firmware #429: Feeds of OpenWrt instead of Lede

niccokunzmann, 2017-03-06:
I recognized that the current version of the firmware uses LEDE but some of the [feeds](https://github.com/freifunk-berlin/firmware/blob/master/feeds.conf#L1) still use OpenWrt.

I assume, they should be moved, too. Is that so?

SvenRoederer, 2017-03-06:
check within LEDE: https://git.lede-project.org/?p=source.git;a=blob;f=feeds.conf.default;h=040dbdc551f209277778336b422e70b3b168cfe0;hb=HEAD

niccokunzmann, 2017-03-06:
It looks like they are the same.

niccokunzmann, 2017-03-06:
Should I create a pull-request?

SvenRoederer, 2017-03-06:
correct, feeds are shared between both projects.
A PR for what? Do you miss a special feature, that was introduced recently?

niccokunzmann, 2017-03-06:
I thought that since the firmware runs on LEDE now, some inconsistencies may be there and should have to be ported. Since this is none of them, I close the issue. Thanks for the clarification!

closed 2017-03-06 by niccokunzmann