freifunk-berlin/firmware #432: Generic MAC80211 802.11bgn, Failed to initialize interface

everloop2, 2017-03-09:
- think its not Freifunk related due to new options at network > wireless
- occured after adding a private wifi only effecting it
- wireless.@wifi-iface[2].**ieee80211w='0'** > breaks whole wifi
- deleting it > wifi working again

Hostname: cottbus-lausi36
Model: Arcor 803 (Easybox803, https://wiki.openwrt.org/toh/astoria/arv752dpw22)
Firmware Version: Freifunk Berlin Hedy 1.0.0-olsrd0903-alpha 9a4fd73 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)
Kernel Version: 4.4.50

Syslog:
daemon.notice netifd: radio0 (1274): command failed: Not supported (-122)
daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Thu Mar  9 13:38:10 2017 daemon.err hostapd: Line 48: unknown configuration item 'ieee80211w'
Thu Mar  9 13:38:10 2017 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Thu Mar  9 13:38:10 2017 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Thu Mar  9 13:38:10 2017 daemon.err hostapd: Failed to initialize interface

booo, 2017-03-13:
You used the administration interface for wireless networks to add a new wireless network? If that's the case I suggest that you report this problem upstream: https://lede-project.org/bugs It may help to create a minimal example for the "bug" based on a standard lede image.

everloop2, 2017-03-17:
Yes, i used administration interface for wireless networks to add a new wireless network.

Only found https://dev.openwrt.org/ticket/20179 searching: "hostapd" and "ieee80211w", noting at LEDE.

I'll test LEDE trunk and then report.

everloop2, 2017-03-19:
Have a build running, using custom: /firmware/feeds.conf

src-git packages https://git.lede-project.org/feed/packages.git
src-git luci https://git.lede-project.org/project/luci.git
src-git routing https://git.lede-project.org/feed/routing.git
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git

Set up private Wifi using FFWizard and afterwards also tried administration interface for wireless networks adding a new wireless network > worked. Seems to be fixed :D

SvenRoederer, 2017-03-19:
so you build against head of all feeds, but the next release will be based on LEDE-release 17.01. 
You tested initially with development-branch "SAm0815/Hedy-olsr0903", which don't look like a current build. 
Can you check with our master? To confirm that the bug is not in 17.01-branch anymore?

everloop2, 2017-03-21:
adding a private AP in Luci > wireless still breaks due option > `wireless.@wifi-iface[2].ieee80211w='0'`
tested https://github.com/freifunk-berlin > branch: master
Firmware: Freifunk Berlin Hedy 0.3.0-alpha e7b8a36 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)

feeds.conf
```
src-git packages https://github.com/openwrt/packages.git^ed90827282851ad93294e370860320f1af428bb2
src-git luci https://github.com/openwrt/luci.git^a100738163585ae1edc24d832ca9bef1f34beef0
src-git routing https://github.com/openwrt-routing/packages.git^dd36dd47bbd75defcb3c517cafe7a19ee425f0af
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git^3186386056cf52ebea3c7298b0a57fa8eafc851e
```

Syslog:
```
daemon.notice netifd: radio0 (5563): command failed: Not supported (-122)
Tue Mar 21 11:20:46 2017 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Tue Mar 21 11:20:46 2017 daemon.err hostapd: Line 46: unknown configuration item 'ieee80211w'
Tue Mar 21 11:20:46 2017 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Tue Mar 21 11:20:46 2017 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Tue Mar 21 11:20:46 2017 daemon.err hostapd: Failed to initialize interface
Tue Mar 21 11:20:47 2017 daemon.notice netifd: radio0 (5563): Device setup failed: HOSTAPD_START_FAILED
```

New Option at LuCI > Network > Wireless > Add > Wireless Security > Encryption > e.g. WPA2-PSK:

802.11w Management Frame Protection > Default > Disabled
help: Requires the 'full' version of wpad/hostapd and support from the wifi driver
(as of Feb 2017: ath9k and ath10k, in LEDE also mwlwifi and mt76)

```
opkg list-installed
hostapd-common - 2016-12-19-ad02e79d-2
wpad - 2016-12-19-ad02e79d-2
```

SvenRoederer, 2017-03-23:
please try currently building job#176 or job#178 from buildboot (branch "SAm0815_Hedy-alpha_olsr0903")

everloop2, 2017-03-25:
Solved. :D

no OWM at moment, fails due https://github.com/freifunk-berlin/firmware/issues/431 & https://github.com/freifunk-berlin/firmware/issues/435

Test Build up and running: http://monitor.berlin.freifunk.net/host.php?h=cottbus-lausi36

Firmware: Freifunk Berlin Hedy 1.0.0-olsrd0903-alpha fbbcf94 r2993+293-b9a408c / LuCI lede-17.01 branch (git-17.073.42825-b47a21f)

feeds.conf
```
src-git packages https://github.com/openwrt/packages.git^3c36d4388cfdb405d423bb0d885f693341484b55
src-git luci https://github.com/openwrt/luci.git^b47a21fbbb8c52ff54e99cb1fc68b11c04070d0d
src-git routing https://github.com/SvenRoederer/openwrt-routing.git;lede-17.01_olsrd0903
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git;SAm0815/integrate
```

SvenRoederer, 2017-03-28:
as soon as #450 is solved, we can update the feeds in master

SvenRoederer, 2017-04-18:
closed by eb1767c, which includes the listed revisions of openwrt-packages and openwrt-luci

closed 2017-04-18 by SvenRoederer