freifunk-berlin/firmware #435: owm.lua doesn't work on lede build

bobster-galore, 2017-03-19:
Since data structure changed owm.lua doesn't work anymore.

SvenRoederer, 2017-03-19:
the way of gathering the data seems to be different now. see https://github.com/openwrt/luci/blob/fecf6e1d11d466efa7b1bbf05633931a3081517e/modules/luci-base/luasrc/tools/status.lua#L160

SvenRoederer, 2017-04-15:
just made branch "https://github.com/freifunk-berlin/firmware-packages/tree/owm-wifi_hotfix" which disables the broken section.
This will be included into all SAm0815* images for now

bobster-galore, 2017-06-13:
There was just an issue [1] on Berlin ml, that an CPE210 1.1 EU isn't showing up in the wifimap using the Development Master (293). There is still an issue with owm.lua and I don't have the time to fix it now.
I propose that we include above mentioned owm-wifi_hotfix in Development Master.
[1] https://lists.berlin.freifunk.net/pipermail/berlin/2017-June/035882.html

SvenRoederer, 2017-06-13:
I just created a RP https://github.com/freifunk-berlin/firmware-packages/pull/108 to merge this hotfix (disabling incompatible code) ...
 Let's discuss on the next community-meeting

booo, 2017-06-15:
This affects the current master branch?

closed 2017-07-02 by SvenRoederer

bobster-galore, 2017-07-26:
Hej folks,
on a wr842v3 running "Freifunk Berlin Hedy SAm0815-ffuplink 5799d8d r2993+469-b9a408c" owm.lua does nothing. It doesn't update a doc nor does it give an error message.
ru bob

SvenRoederer, 2017-07-26:
What is "owm.lua --dry-run" giving as output?

bobster-galore, 2017-07-26:
errr, didn't save the original owm.lua when testing different versions, now it's from Kathleen 0.3.0.
And working, before I tried --dryrun. :( Can't there be a hint to --dry-run?
So the coordinates were missing despite I entered them in the wizard.

booo, 2017-07-26:
Please open another ticket for this problem. We definitely can work on this.

bobster-galore, 2017-07-27:
How u mean?
hint 4 --dry-run
or 
entered coordinates disappear (will check w/ firstboot when time permits.)

booo, 2017-07-27:
> "Can't there be a hint to --dry-run?"

Maybe I misinterpreted you writing but I thought you requested a bit more output from owm.lua, e.g. a --help argument or a message when the coordinates are missing.

bobster-galore, 2017-09-22:
U were right and I opened #485.