freifunk-berlin/firmware #437: ff-assistent: shows textbox with OpenVPN-key paths inside

SvenRoederer, 2017-03-19:
with Updated LuCI of 2017 the ffwizard shows the following screen when uploading the VPN-keys
[image]

These Inputboxes should be invisible as in previous releases

SvenRoederer, 2017-10-09:
I feel like I saw this in the "luci-app-openvpn" also. So this seems a generic behavior and not considered as a bug.
Any opinions?

bobster-galore, 2017-10-18:
It seems to be an upstream problem. We watch it?! postponed to 1.1.0?

SvenRoederer, 2017-10-19:
these additional textboxes can also bee seen in the luci-app-openvpn page.
So it might be a wanted thing from upstream.
[image]

bobster-galore, 2017-10-20:
I postpone it until we drop it.

SvenRoederer, 2017-10-28:
yeah, let's wait for users feedback with Hedy-1.0.0 on this and then decide if we just close or a change is required

closed 2018-09-30 by bobster-galore