freifunk-berlin/firmware #438: ffwizard: Community is not preset to "Freifunk Berlin" with 2017-LuCI

SvenRoederer, 2017-03-19:
on the 1st page of the Wizard the "Freifunk Berlin" community should be preselected. Instead the 1 listitem is choosen.

booo, 2017-06-25:
As far as I understand the situation the /etc/config/freifunk configuration file gets not properly created e.g. from the /etc/config/profile_berlin which leads to this and other problems (5ghz wifi?).

Does someone understand where we create the /etc/config/freifunk file or update with data from the community profile?

booo, 2017-06-25:
Seems like two packages try to create the same file:

https://github.com/freifunk-berlin/firmware-packages/blob/master/defaults/freifunk-berlin-freifunk-defaults/files/etc/config/freifunk

https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-common/files/etc/config/freifunk

The Makefile creates the files with a `$(CP) command`.

The freifunk-common package is a dependency of the community-profiles package.

We could change our defaults package to a uci-default script and circumvent the problem this way.

SvenRoederer, 2017-07-10:
setting the correct community in etc/config/freifunk will cause the correct preselection. 
so https://github.com/freifunk-berlin/firmware-packages/pull/113 will fix this

closed 2017-07-10 by SvenRoederer