freifunk-berlin/firmware #440: ffwizard: OpenVPN .crt & .key get in wrong place + wrong name

everloop2, 2017-03-21:
branch: master
firmware: Freifunk Berlin Hedy 0.3.0-alpha e7b8a36 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)
target: lantiq > x-way > Easybox803a

feeds.conf
```
src-git packages https://github.com/openwrt/packages.git^ed90827282851ad93294e370860320f1af428bb2
src-git luci https://github.com/openwrt/luci.git^a100738163585ae1edc24d832ca9bef1f34beef0
src-git routing https://github.com/openwrt-routing/packages.git^dd36dd47bbd75defcb3c517cafe7a19ee425f0af
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git^3186386056cf52ebea3c7298b0a57fa8eafc851e
```
added .crt & .key in ffwizard > finished wizard and reboot

Syslog:
`Tue Mar 21 10:47:26 2017 daemon.err openvpn(ffvpn)[4192]: Cannot load certificate file /etc/openvpn/freifunk_client.crt`

downloaded configuration backup archive via LuCI > found
.crt in /etc/luci-uploads/cbid.ffvpn.1.cert
.key in /etc/luci-uploads/cbid.ffvpn.1.key

if uploaded manually via LuCI > Services > OpenVPN.. files default to:
```
/etc/luci-uploads/cbid.openvpn.ffvpn.ca
/etc/luci-uploads/cbid.openvpn.ffvpn.cert
/etc/luci-uploads/cbid.openvpn.ffvpn.key
```
then OpenVPN is working

SvenRoederer, 2017-03-21:
isn't this a duplicate of #421 ?

If so branches "SAm0815/Hedy-alpha" should have a fix (https://github.com/freifunk-berlin/firmware-packages/pull/103)

SvenRoederer, 2017-03-21:
please reopen in case it is not

closed 2017-03-21 by SvenRoederer

everloop2, 2017-03-25:
was duplicate 

Solved in SAm0815_Hedy-alpha_olsr0903 :D

Firmware: Freifunk Berlin Hedy 1.0.0-olsrd0903-alpha fbbcf94 r2993+293-b9a408c / LuCI lede-17.01 branch (git-17.073.42825-b47a21f)

feeds.conf
```
src-git packages https://github.com/openwrt/packages.git^3c36d4388cfdb405d423bb0d885f693341484b55
src-git luci https://github.com/openwrt/luci.git^b47a21fbbb8c52ff54e99cb1fc68b11c04070d0d
src-git routing https://github.com/SvenRoederer/openwrt-routing.git;lede-17.01_olsrd0903
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git;SAm0815/integrate
```