freifunk-berlin/firmware #441: Lede Build - 5Ghz not enabled

pmelange, 2017-03-24:
After running the wizard on a fresh install, the 5Ghz radio is not enabled.

Target platform: wdr3600
Build: Freifunk Berlin Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

booo, 2017-06-18:
I can confirm this problem.

I will try to take a look and maybe move the wifi specific code out of the wizards.

booo, 2017-06-20:
Inspiration for reworking the wifi setup code:

https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/15-wireless.sh

https://github.com/freifunk-berlin/firmware/issues/293

Maybe we should rework the upstream code so we can use it as library for our setup process:

https://github.com/lede-project/source/blob/079d57b0f290a79c9dbc013b6e9c83cebf8a2f99/package/kernel/mac80211/files/lib/wifi/mac80211.sh

booo, 2017-06-25:
I suggest that we merge the referenced pull request and rework the wifi setup code another day.

booo, 2017-06-25:
Actually this is caused by the freifunk config creation problem described in this ticket: https://github.com/freifunk-berlin/firmware/issues/438

booo, 2017-08-11:
Should work again.

closed 2017-08-11 by booo